pixelit-project/PixelIt #72: Garbled text problem scrollText

fmunozs, 2021-09-16:
When using the test functionality I see garbled text  sometimes on the matrix when trying to draw some short text (when no scrolling is needed). This happens with the default test, which sets the scrollText to auto.

`connection.send('{"text":{"textString":"hello","scrollText":"auto","scrollTextDelay":45,"bigFont":false,"centerText":false,"position":{"x":0,"y":1},"color":{"r":255,"g":255,"b":255}}}')
`
if I switch scrollText to false, it always works fine.
`connection.send('{"text":{"textString":"hello","scrollText":"false","scrollTextDelay":45,"bigFont":false,"centerText":false,"position":{"x":0,"y":1},"color":{"r":255,"g":255,"b":255}}}')`

o0shojo0o, 2021-09-17:
I cannot confirm this behavior. 
Which version do you have installed?

fmunozs, 2021-09-17:
I'm using firmware_d1_mini.bin 0.3.1 on a wemos d1 clone. I'll share a quick video in a minute. Everything else seems to be working regarding drawing text

fmunozs, 2021-09-17:
I'm sending the text "id", sometimes it's drawn broken, and sometimes it works ok. then it shows the time, and I send the text "idea".

File is too big to attach to Github issue., I've attached stills from the video 

[image]

Can you please try to download the video from here? 

`https://www.icloud.com/attachment/?u=https%3A%2F%2Fcvws.icloud-content.com%2FB%2FAaTBTDzwbPj3RBBKeKN9fwb2MVzFAW1NkPUCgUe4Tte2aZEDSJEScIwH%2F%24%7Bf%7D%3Fo%3DAk8w3N-0ephSA_xZs7_8oKrlAYxTLnTGyQIWAZLF_YRA%26v%3D1%26x%3D3%26a%3DCAoghvNVab1WMvSP-cXkYrZyvR-K4q6erOQoww6uKuP7gucSeBCI-eejvy8YiInj98gvIgEAKgkC6AMA_0XIB7lSBPYxXMVaBBJwjAdqJik_QF-W1X-_e7CLxHTeBFr28RmMC96yX5VIAi4j-nBhmf0ujtF8ciZx1WjkYvY2uljjvUjI09yBp3KHAleQLAMGTPDXFueVpO8pJYwd-w%26e%3D1634486240%26fl%3D%26r%3D3F501BD6-D379-486E-B17F-DD816B8E1B30-1%26k%3D%24%7Buk%7D%26ckc%3Dcom.apple.largeattachment%26ckz%3DB85C552F-68F0-459C-9A9E-24B9B3898761%26p%3D45%26s%3D-maiNBwBx8PNJo9H6navHjxoPHI&uk=fSSBQDhKhIsEffrEp61gnQ&f=IMG_8178.MOV&sz=58802468`

o0shojo0o, 2021-09-17:
Are you really using version 0.3.1?
Because we are now at version 0.3.10.

fmunozs, 2021-09-17:
Sorry, You are right, it's definitely 0.3.10!
[image]

o0shojo0o, 2021-09-17:
I have now used the JSON skill about 50 times, without this error.

```json
{"text":{"textString":"id","scrollText":"auto","scrollTextDelay":45,"bigFont":false,"centerText":false,"position":{"x":0,"y":1},"color":{"r":255,"g":255,"b":255}}}
```

What you could try once with the CleanFlash.bin (is included in the download of the Flasher) to clean the flash once, and then fresh reflash.
https://docs.bastelbunker.de/pixelit/firmware.html#flashen

fmunozs, 2021-09-18:
I tried cleanflashing and reflashing again, samething happens. I'll get a different mini d1 and let you know if it also happens. thanks.