pixelit-project/PixelIt #77: Ability to change color order

TravisRoy, 2021-10-04:
My LEDs seem to have a different color order than what PixelIt is sending. WLED has the option to change it.
[image]

o0shojo0o, 2021-10-05:
I had a quick look yesterday, it should be possible. 

As soon as I'm done with the new UI I'll take a closer look at this topic.

However, this may take some time because I currently can not spend much time.

o0shojo0o, 2021-10-16:
Unfortunately I was too optimistic here, I use FastLED to control the LEDs, this lib does not allow me to set the color order at runtime (https://github.com/FastLED/FastLED/issues/282#issuecomment-744562566).

I have to compile every color order and the memory of the ESP8266 doesn't support that, it's just too small.

o0shojo0o, 2021-10-17:
However, you can create a custom version where you only need to adjust the code in three places.
https://github.com/o0shojo0o/PixelIt/blob/c3429897945d4815e5970459dc6e1c4cc014d7b5/src/PixelIt.ino#L1990-L2003

For **GRB** it would look like this: 
```cpp
 // Matrix Color Correction 
 if (userLEDCorrection != UncorrectedColor) 
 { 
 	FastLED.addLeds<WS2812B, MATRIX_PIN, GRB>(leds, NUMMATRIX).setCorrection(userLEDCorrection); 
 } 
 else if (userColorTemp != UncorrectedTemperature) 
 { 
 	FastLED.addLeds<WS2812B, MATRIX_PIN, GRB>(leds, NUMMATRIX).setTemperature(userColorTemp); 
 } 
 else 
 { 
 	int *rgbArray = GetUserCutomCorrection(); 
 	FastLED.addLeds<WS2812B, MATRIX_PIN, GRB>(leds, NUMMATRIX).setCorrection(matrix->Color(rgbArray[0], rgbArray[1], rgbArray[2])); 
 } 
```

o0shojo0o, 2021-10-17:
Which color order do you need?

TravisRoy, 2022-06-27:
I'm pretty sure I need the WS2812B order