pixelit-project/PixelIt #79: Selectable Pins for LED Disp and Sensors - Enhancement

rellis-erigon, 2021-10-17:
A+ for the recent Updats. Is it possible to have a selective Pin for the LED Display and othe sensors. I note the common pin for the Wemos D1 Mini shield uses D4 the same as the LED display currently. https://hobbycomponents.com/shields/868-wemos-d1-mini-dht-temphum-shield.

miccgn, 2021-12-01:
+1 from me as well. I'd love to create an option which allows to shift the D8 pin to D5, as this would allow to re-use components from AWTRIX project (AWTRIX PCB expects DFPlayer on D5).
I am eager to extend the .ino code, however, I have no idea how to extend the WebUi.

miccgn, 2021-12-03:
I started modifying the source in order to make the pin assignments dynamic. I made quite a lot progress for the sensors. However, when I came across the matrix pin, I had to learn that this cannot be done dynamically on a ESP8266.

The following discussion explains the reasons behind this:
https://github.com/FastLED/FastLED/issues/1137
To summarize: in order to gain the required speed even on an ESP8266, the FastLED library tries to make as little decisions during runtime as possible. One of these decisions is the pin used. **The pin can only be changed when the software is being compiled.** 

The only option would be to create as many FastLED objects within the program as pins should be available, then decide dynamically which one to use. Disadvantage of that solution: it consumes quite a lot memory.

Given that @o0shojo0o even moved the webui to github rather than keeping it on flash and in memory, I suppose memory will be an issue.

So, I will try to finish my work on dynamic pins **for sensors** during the next couple of days and upload the code as PR, which allows @o0shojo0o to decide whether to accept my modifications or not. But LEDs will remain assigned to D4.

miccgn, 2021-12-04:
I have finished working on the pin assignment version.
However, I think I might have mixed up my repository, basing the new version on 0.3.14 then merging and reverting, rather than basing it on my recent PR.
@o0shojo0o I guess it would be easiest if you could decide on my recent PR, then I will create a new PR for the pin assignment version.

PR #94 (2021-12-12): Sensor pins assignable; LDR options; bugfix
* (ESP8266 only) The pins for sensors can be chosen in the options and will be stored in the configuration file.
* LDR type and pulldown resistor value can be set in the options.
* MP3Player sometimes cannot handle two commands in a row ("volume", then "play"). Insert delay if there's a second command.
Partially resolves #79 and #17 .
closed 2021-12-12 by o0shojo0o