pixelit-project/PixelIt #85: Temperature/Humidity stopped working after update to 0.3.6

hamster65, 2021-11-07:
Hi,
since release 0.3.6 when support for BME280 was added, my sensor AM2320 returns "not installed" most of the time. 

I tried to figure out what is wrong and discovered that it works after I reset the ESP (for example by klicking "save config" in the webinterface). It does not work after a power cycle.

Maybe a delay is needed after power up before initialization?  I guess the DHT22 does not need that delay which is why no one else seems to be having that issue :)

o0shojo0o, 2021-11-10:
Hi, 
I'll take a look, however I don't have an AM2320 here.

hamster65, 2021-11-10:
Hi,
I am testing right now. The sensor supports i2C and single bus mode. For single bus mode, which is the same as with DHT22,  SCL (connected to GND) must be low for 500 ms. I guess reading the sensor earlier puts it into I2C mode. The datasheet also says that only a power cycle can change the mode. Page 5:
https://cdn-shop.adafruit.com/product-files/3721/AM2320.pdf

I tested a delay right before dht.setup. 400ms was too short, 500 worked most of the time, so I am using 600 ms now.

o0shojo0o, 2021-11-12:
Hi, I think we can work with a delay here, the 600 ms should not be a problem.

hamster65, 2021-11-13:
Cool, thanks! 600ms works here without any problems so far.

o0shojo0o, 2021-11-14:
fixed in  v0.3.14

hamster65, 2022-01-25:
Hi, after flashing the latest version, my sensor stoppend working. I think this happens because it checks for other sensors on the same pin before the 600 ms delay. 

I have not verified this because I am missing getSmoothedLux. How do I update the library? I expected vscode / platform.io to do this automatically.

Thanks!

hamster65, 2022-01-25:
Got it, I had to do a "clean all" to get the latest version of the LDR lib. I moved the delay up to where it says "// Init Temp Sensors" and it's working again.

hamster65, 2022-02-06:
Seems like a longer delay of 800ms is needed to make it work reliably.
The problem that the sensor could not be detected happend only in the morning after powering up my Pixelit. Power cycles during the day, leaving it off for some minutes, resulted in a working sensor. Perhaps it depends on the room temperature.

closed 2022-03-15 by o0shojo0o