pixelit-project/PixelIt #96: Combine centerText with scrollText=auto

miccgn, 2021-12-10:
It would be a nice enhancement if I could specify both "centerText=true" and "scrollText=auto".

Expected behaviour:
- if text fits into display: center it, no scrolling
- if text does not fit into display: ignore centering, start scrolling

Difference to now is that text that fits will be centered, not adjusted left.

I have some ideas how to do this, but this requires redesign of all the DrawText... functions.

o0shojo0o, 2021-12-15:
Yes, it would have to completely rebuild DrawText.
But I couldn't motivate myself to do it yet. 
I had started it once before, but discarded it due to lack of time.

miccgn, 2023-09-05:
Still would be an useful extension

foorschtbar, 2023-09-11:
@miccgn please test the latest beta: https://github.com/pixelit-project/PixelIt/releases/tag/2.3.2-beta

foorschtbar, 2023-09-23:
Fixed with latest (2.3.2) release

closed 2023-09-23 by foorschtbar

miccgn, 2023-09-23:
Thanks!
Sorry, didn't find time to test it yet 😕😕