pixelit-project/PixelIt #97: Save config button greyed out ESP01

Zamboca, 2021-12-16:
Hi; 
Im not so sure if this is the correct spot to post my issue, and if not please advise and i will repost. Im using [ESP01](https://www.amazon.ca/ESP8266-Wireless-Transceiver-Receiver-Raspberry/dp/B09DD1N7J1/ref=sr_1_4_sspa?keywords=esp01&qid=1639684787&sr=8-4-spons&psc=1&spLa=ZW5jcnlwdGVkUXVhbGlmaWVyPUEzNTZTVTUwQTY3VjhLJmVuY3J5cHRlZElkPUEwNTg2NjYwMzZXVjhFMVczUVRGWCZlbmNyeXB0ZWRBZElkPUEwMTE5NDcyMTlDMUI5TUg3TFVCRCZ3aWRnZXROYW1lPXNwX2F0ZiZhY3Rpb249Y2xpY2tSZWRpcmVjdCZkb05vdExvZ0NsaWNrPXRydWU=) which is based on the 8266. The firmware loaded successfully but i cant seem to save any configuration. Could it be that this firmware file i used doesnt work with ESP01 ? ( I faced a similar issue with WLED untill i loaded the correct .bin )
Any help is greatly appreciated

[image]

o0shojo0o, 2021-12-17:
Hi, 
it looks like not even the defaults are loaded. 
Therefore I assume that the used ESP01 uses a different memory layout. 

Which binary did you use in the case?

Zamboca, 2021-12-17:
This screenshot was taken when i loaded firmware_esp8266.bin . I tried loading the firmware_d1_mini.bin, but when i booted up i didnt get the pixelit startup, and it was the same result and cant save configs. Thank you

o0shojo0o, 2022-01-02:
Sorry, the ESP01 is not supported.
First a device configuration in the [platformio.ini](https://github.com/o0shojo0o/PixelIt/blob/master/platformio.ini) for the ESP01 and test it.

closed 2022-01-02 by o0shojo0o