pixelit-project/PixelIt #100: Jumping LDR readings

miccgn, 2021-12-19:
Duplicating https://github.com/QuentinCG/Arduino-Light-Dependent-Resistor-Library/issues/7#issue-1084082122 - it could be solved there or here, waiting for the library's author to decide:

when I built a device with a display that was expected to adapt to ambient lighting, I noticed the readings of my GL5528 jumping around as soon as ceiling lighting was powered on. This lighting consists of LEDs controlled by phase-cut dimmers.

It seems that the LDR, which has a top speed of 20ms (==50 readings per second) was sensitive to this kind of dimmers. There hadn't been any "jumping values" with classic halogene bulbs.

So I was wondering whether it would be a good idea to implement some kind of smoothing ~~within the library, offering a parameter specifiying the number of past readings to be used for smoothing~~?

miccgn, 2021-12-19:
(Of course I will contribute and implement the smoothing, if accepted)

miccgn, 2021-12-20:
I created a PR for the LDR library.

Nevertheless, my problem still persists. As this definitely isn't a software issue, I'll close this "Issue" and create a discussion.

miccgn, 2021-12-20:
I created a PR for the LDR library. After the owner of the library accepts the PR, @o0shojo0o would need to 
- add a parameter in the photocell constructor specifying the number of readings to be used for smoothing (I'd recommend at least 10, which would smooth along 10 readings, taking 10 seconds to fully adapt for permanent changes, but maybe even 20 or 30).
- replace calls to getCurrentLux() by getSmoothedLux()

miccgn, 2021-12-20:
I still have trouble with my LDR, but will continue this in a discussion.

hamster65, 2021-12-20:
Fluctuations could be because of unstable supply / voltage drop or unstable voltage reference of the ESP.

Mine is mostly stable, but sometimes PixelIt changes back and forth between two brighness levels. I did some smoothing in node-red before the brighness regulation feature got built in.

miccgn, 2021-12-24:
Added BH1750 support, which partly mitigates this issue.

I'll keep this issue open, as smoothing still might be helpful (only available with LDR, not with BH1750).

miccgn, 2022-01-08:
Solved by 0.3.16

closed 2022-01-08 by miccgn