pixelit-project/PixelIt #112: Web dashboard not loading

RiggsUS, 2022-01-02:
I can seem to load the webui from safari or Firefox on MacOS, the only way to load it is from Edge on Windows 11.
I do have 1Blocker and tried to disable it and reload the UI
Errors on Safari:
Cross-origin script load denied by Cross-Origin Resource Sharing policy.
1Blocker | JXH – "Not found corresponding domains for hostname \"pixelit.local\""
error
(anonymous function) — Script Element 6:1:2956
asyncFunctionResume
(anonymous function)
promiseReactionJobWithoutPromise
promiseReactionJob
Script Element 6:1:2421

o0shojo0o, 2022-01-02:
Already tried to call the Pixel it via the ip?

RiggsUS, 2022-01-02:
Sorry yes I did, here are the errors:
[Error] 1Blocker | 54B – "Not found corresponding domains for hostname \"192.168.1.193\""
	error
	(anonymous function) (Script Element 6:1:2956)
	asyncFunctionResume
	(anonymous function)
	promiseReactionJobWithoutPromise
	promiseReactionJob
[Error] Failed to load resource: too many HTTP redirects (app.879fa878.css, line 0)
[Error] Failed to load resource: too many HTTP redirects (chunk-vendors.e894fa67.css, line 0)
[Error] Failed to load resource: too many HTTP redirects (chunk-vendors.36a3f74f.js, line 0)
[Error] Failed to load resource: too many HTTP redirects (app.0faf8078.js, line 0)

o0shojo0o, 2022-01-02:
Sorry I don't think there is much I can do, it seems that the 1Blocker is a bit too eager.

RiggsUS, 2022-01-02:
The above errors were with 1Blocker disabled and also with the pages added to the whitelist.
now this is what I see from FireFox on MacOS, with no extensions installed:
Layout was forced before the page was fully loaded. If stylesheets are not yet loaded this may cause a flash of unstyled content. 192.168.1.193
Loading failed for the <script> with source “https://o0shojo0o.github.io/PixelIt/new_webif/js/chunk-vendors.36a3f74f.js”. 192.168.1.193:1:1
The resource at “https://o0shojo0o.github.io/PixelIt/new_webif/js/app.0faf8078.js” preloaded with link preload was not used within a few seconds. Make sure all attributes of the preload tag are set correctly. 192.168.1.193

o0shojo0o, 2022-01-03:
I do not get the behavior provoked. 
Tested with Chrome, Firefox and Edge on Windows 11 and Safari on iPhone and iPad. 

Also I have no influence on the order how the js or css are loaded, because this is a vue project.

Also it can't be because of webspace which is github-pages.
(https://github.com/o0shojo0o/PixelIt/tree/master/docs/new_webif)

RiggsUS, 2022-01-04:
No problem thanks for taking a look, ill keep playing around and see if I can work it out.

closed 2022-01-04 by RiggsUS