pixelit-project/PixelIt #127: BUG: Webinterface not loading on Safari

Kugelfang666, 2022-01-26:
I'm encountering an issue where the PixelIT web interface does not load on Safari (iPad or Mac). Clearing the cache helps fixing the issue temporarily. When I checked the console I found this:

[image]

Other browsers like chrome do not encounter this problem. I'm running he lasted FW 0.3.16

o0shojo0o, 2022-02-02:
I only have iPhones here at hand and they do not cause any problems. 
However, this problem seems to occur only on Apple devices.

rliegmann, 2022-02-02:
I have a Mac with Safari and there I don't get a single error message in the console.
I just googled the error message. Maybe this helps [here](https://support.apple.com/de-de/HT203370).

o0shojo0o, 2022-05-13:
I have now changed the DNS entries and "Optimized", I hope that now the problem is solved.

closed 2022-05-13 by o0shojo0o

Kugelfang666, 2022-05-13:
sorry I must have overlooked this, will test and report back if this issue is solved now