pixelit-project/PixelIt #128: Issues compiling with es32 d1 mini

diogopms, 2022-01-26:
Issues compiling with es32 d1 mini

diogopms, 2022-01-26:
Update: If i comment the Wifimanager i was able to enter in the dashboard.

[image]

o0shojo0o, 2022-02-02:
Again, I can't say much about this as I don't use esp32 .
@rliegmann maybe?

rliegmann, 2022-02-02:
Hey diogopms,
I'll take a look right away.
I was even able to reproduce the error. I'll report when I have something new.
LG rliegmann

rliegmann, 2022-02-02:
UPDATE:
For a quick fix or test, can you please insert the following line in front of the WifiManager and transfer it?

`WiFi.mode(WIFI_AP);`

I also noticed that the WiFi stuff doesn't work properly when using the serial port via USB. So flash it and then operate it on a USB power supply and see if the WiFi shows up.
If that works, I would adjust the project configuration file for your board.

LG eliegmann

closed 2022-05-22 by o0shojo0o