pixelit-project/PixelIt #129: Save config is not working for ESP32

diogopms, 2022-01-26:
[image]

```sh
/#/:1 Uncaught SyntaxError: Unexpected token } in JSON at position 10
    at JSON.parse (<anonymous>)
    at t.value (build.js:1)
    at t.value (build.js:1)
    at WebSocket.forEach.t.WebSocket.<computed> (build.js:1)
```

o0shojo0o, 2022-02-02:
Sorry can not say much about it, I do not use esp32.
@rliegmann can you say something about it?

rliegmann, 2022-02-02:
Yes, I might be able to help here too. I had that problem a long time ago too. 
The whole thing has something to do with the SPIFFS memory. For some reason the ESP cannot initialize the memory properly. I could solve the problem by simply transferring the example Scatch (SPIFFS_Test.ino) from the Arduino library to the ESP. This will take a read and write test. Then simply iron over the Pixel IT firmware again and it should work.

LG rliegmann

closed 2022-05-22 by o0shojo0o