pixelit-project/PixelIt #146: animated bitmaps seem broken from node-red function

poblabs, 2022-02-24:
Hi there, when I use the animated home assistant bitmap below from the test area, it displays perfectly. 

When I try it using the pixel it bitmap animation node, it seems to display fine. 

When I try it from Node-RED function it animates about 100 times faster and it looks bad. In one case it crashed the esp8266 and it rebooted. 

Any thoughts why this would happen?

---

Here's my node-red flow:

[image]

```
[{"id":"9da473733177f0a1","type":"inject","z":"e6d349d60ead1d22","name":"","props":[{"p":"payload"},{"p":"topic","vt":"str"}],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"","payload":"","payloadType":"date","x":880,"y":40,"wires":[["bad72ede352e916e"]]},{"id":"f799b03df876cf50","type":"Bitmap Animation","z":"e6d349d60ead1d22","name":"Bitmap Animation","indata":"[1374,1374,1374,65535,65535,1374,1374,1374,1374,1374,65535,65535,65535,65535,1374,1374,1374,65535,65535,65535,65535,65535,65535,1374,65535,65535,65535,65535,65535,65535,65535,65535,1374,65535,65535,65535,65535,65535,65535,1374,1374,65535,65535,65535,65535,65535,65535,1374,1374,65535,65535,65535,65535,65535,65535,1374,1374,65535,65535,65535,65535,65535,65535,1374],[1374,1374,1374,65535,65535,1374,1374,1374,1374,1374,65535,65535,65535,65535,1374,1374,1374,65535,65535,65535,65535,65535,65535,1374,65535,65535,65535,65535,65535,65535,65535,65535,1374,65535,65535,65535,65535,65535,65535,1374,1374,65535,65535,65535,65535,65535,65535,1374,1374,65535,65535,65535,65535,65535,65535,1374,1374,65535,65535,65535,1374,65535,65535,1374],[1374,1374,1374,65535,65535,1374,1374,1374,1374,1374,65535,65535,65535,65535,1374,1374,1374,65535,65535,65535,65535,65535,65535,1374,65535,65535,65535,65535,65535,65535,65535,65535,1374,65535,65535,65535,65535,65535,65535,1374,1374,65535,65535,65535,65535,65535,65535,1374,1374,65535,65535,1374,65535,65535,65535,1374,1374,65535,65535,65535,1374,65535,65535,1374],[1374,1374,1374,65535,65535,1374,1374,1374,1374,1374,65535,65535,65535,65535,1374,1374,1374,65535,65535,65535,65535,65535,65535,1374,65535,65535,65535,65535,65535,65535,65535,65535,1374,65535,65535,65535,1374,65535,65535,1374,1374,65535,65535,65535,1374,65535,65535,1374,1374,65535,65535,1374,65535,65535,65535,1374,1374,65535,65535,65535,1374,65535,65535,1374],[1374,1374,1374,65535,65535,1374,1374,1374,1374,1374,65535,65535,65535,65535,1374,1374,1374,65535,65535,65535,65535,65535,65535,1374,65535,65535,65535,1374,65535,65535,65535,65535,1374,65535,65535,65535,1374,1374,65535,1374,1374,65535,65535,65535,1374,65535,65535,1374,1374,65535,65535,1374,65535,65535,65535,1374,1374,65535,65535,65535,1374,65535,65535,1374],[1374,1374,1374,65535,65535,1374,1374,1374,1374,1374,65535,65535,65535,65535,1374,1374,1374,65535,65535,65535,1374,65535,65535,1374,65535,65535,65535,1374,65535,65535,65535,65535,1374,65535,1374,65535,1374,1374,65535,1374,1374,65535,65535,65535,1374,65535,65535,1374,1374,65535,65535,1374,65535,65535,65535,1374,1374,65535,65535,65535,1374,65535,65535,1374],[1374,1374,1374,65535,65535,1374,1374,1374,1374,1374,65535,1374,65535,65535,1374,1374,1374,65535,65535,65535,1374,65535,65535,1374,65535,1374,65535,1374,65535,65535,1374,65535,1374,65535,1374,65535,1374,1374,65535,1374,1374,65535,65535,65535,1374,65535,65535,1374,1374,65535,65535,1374,65535,65535,65535,1374,1374,65535,65535,65535,1374,65535,65535,1374]","inanimationDelay":"100","inrubberbanding":"false","inlimitLoops":"0","x":1110,"y":40,"wires":[[]]},{"id":"bad72ede352e916e","type":"function","z":"e6d349d60ead1d22","name":"icon set?","func":"msg.bitmapAnimation = {};\nmsg.bitmapAnimation.data = \"[1374,1374,1374,65535,65535,1374,1374,1374,1374,1374,65535,65535,65535,65535,1374,1374,1374,65535,65535,65535,65535,65535,65535,1374,65535,65535,65535,65535,65535,65535,65535,65535,1374,65535,65535,65535,65535,65535,65535,1374,1374,65535,65535,65535,65535,65535,65535,1374,1374,65535,65535,65535,65535,65535,65535,1374,1374,65535,65535,65535,65535,65535,65535,1374],[1374,1374,1374,65535,65535,1374,1374,1374,1374,1374,65535,65535,65535,65535,1374,1374,1374,65535,65535,65535,65535,65535,65535,1374,65535,65535,65535,65535,65535,65535,65535,65535,1374,65535,65535,65535,65535,65535,65535,1374,1374,65535,65535,65535,65535,65535,65535,1374,1374,65535,65535,65535,65535,65535,65535,1374,1374,65535,65535,65535,1374,65535,65535,1374],[1374,1374,1374,65535,65535,1374,1374,1374,1374,1374,65535,65535,65535,65535,1374,1374,1374,65535,65535,65535,65535,65535,65535,1374,65535,65535,65535,65535,65535,65535,65535,65535,1374,65535,65535,65535,65535,65535,65535,1374,1374,65535,65535,65535,65535,65535,65535,1374,1374,65535,65535,1374,65535,65535,65535,1374,1374,65535,65535,65535,1374,65535,65535,1374],[1374,1374,1374,65535,65535,1374,1374,1374,1374,1374,65535,65535,65535,65535,1374,1374,1374,65535,65535,65535,65535,65535,65535,1374,65535,65535,65535,65535,65535,65535,65535,65535,1374,65535,65535,65535,1374,65535,65535,1374,1374,65535,65535,65535,1374,65535,65535,1374,1374,65535,65535,1374,65535,65535,65535,1374,1374,65535,65535,65535,1374,65535,65535,1374],[1374,1374,1374,65535,65535,1374,1374,1374,1374,1374,65535,65535,65535,65535,1374,1374,1374,65535,65535,65535,65535,65535,65535,1374,65535,65535,65535,1374,65535,65535,65535,65535,1374,65535,65535,65535,1374,1374,65535,1374,1374,65535,65535,65535,1374,65535,65535,1374,1374,65535,65535,1374,65535,65535,65535,1374,1374,65535,65535,65535,1374,65535,65535,1374],[1374,1374,1374,65535,65535,1374,1374,1374,1374,1374,65535,65535,65535,65535,1374,1374,1374,65535,65535,65535,1374,65535,65535,1374,65535,65535,65535,1374,65535,65535,65535,65535,1374,65535,1374,65535,1374,1374,65535,1374,1374,65535,65535,65535,1374,65535,65535,1374,1374,65535,65535,1374,65535,65535,65535,1374,1374,65535,65535,65535,1374,65535,65535,1374],[1374,1374,1374,65535,65535,1374,1374,1374,1374,1374,65535,1374,65535,65535,1374,1374,1374,65535,65535,65535,1374,65535,65535,1374,65535,1374,65535,1374,65535,65535,1374,65535,1374,65535,1374,65535,1374,1374,65535,1374,1374,65535,65535,65535,1374,65535,65535,1374,1374,65535,65535,1374,65535,65535,65535,1374,1374,65535,65535,65535,1374,65535,65535,1374]\";\n \nreturn msg;","outputs":1,"noerr":0,"initialize":"","finalize":"","libs":[],"x":1300,"y":60,"wires":[["37ec99d6492f2d4b"]]},{"id":"37ec99d6492f2d4b","type":"Screen Init","z":"e6d349d60ead1d22","name":"Screen Init","inscreenName":"test","induration":"10","x":1470,"y":40,"wires":[["212df7a3bec392af"]]},{"id":"212df7a3bec392af","type":"Switch Animation","z":"e6d349d60ead1d22","name":"Switch Animation","aktiv":"false","animation":"fade","x":1650,"y":40,"wires":[["9ce538627d4e60b8"]]},{"id":"9ce538627d4e60b8","type":"Text","z":"e6d349d60ead1d22","name":"Text","intextString":"TESTING BITMAP","inbigFont":"false","inscrollText":"true","inscrollTextDelay":"0","incenterText":"false","inposX":"8","inposY":"1","incolorR":"255","incolorG":"255","incolorB":"255","x":1810,"y":40,"wires":[["2ae8103e.2bfa8"]]},{"id":"2ae8103e.2bfa8","type":"Alert Screen","z":"e6d349d60ead1d22","name":"Alert Screen","x":1570,"y":280,"wires":[["35ecb3fe.70a82c","b3601a19c326a8d6"]]},{"id":"b3601a19c326a8d6","type":"Core","z":"e6d349d60ead1d22","name":"Pixelit Large Core","ip":"192.168.0.159","masterTopic":"","x":1850,"y":320,"wires":[[]]}]
```

---

The bitmap itself:

```
[1374,1374,1374,65535,65535,1374,1374,1374,1374,1374,65535,65535,65535,65535,1374,1374,1374,65535,65535,65535,65535,65535,65535,1374,65535,65535,65535,65535,65535,65535,65535,65535,1374,65535,65535,65535,65535,65535,65535,1374,1374,65535,65535,65535,65535,65535,65535,1374,1374,65535,65535,65535,65535,65535,65535,1374,1374,65535,65535,65535,65535,65535,65535,1374],[1374,1374,1374,65535,65535,1374,1374,1374,1374,1374,65535,65535,65535,65535,1374,1374,1374,65535,65535,65535,65535,65535,65535,1374,65535,65535,65535,65535,65535,65535,65535,65535,1374,65535,65535,65535,65535,65535,65535,1374,1374,65535,65535,65535,65535,65535,65535,1374,1374,65535,65535,65535,65535,65535,65535,1374,1374,65535,65535,65535,1374,65535,65535,1374],[1374,1374,1374,65535,65535,1374,1374,1374,1374,1374,65535,65535,65535,65535,1374,1374,1374,65535,65535,65535,65535,65535,65535,1374,65535,65535,65535,65535,65535,65535,65535,65535,1374,65535,65535,65535,65535,65535,65535,1374,1374,65535,65535,65535,65535,65535,65535,1374,1374,65535,65535,1374,65535,65535,65535,1374,1374,65535,65535,65535,1374,65535,65535,1374],[1374,1374,1374,65535,65535,1374,1374,1374,1374,1374,65535,65535,65535,65535,1374,1374,1374,65535,65535,65535,65535,65535,65535,1374,65 [... cut ...] ,1374,1374,65535,65535,65535,1374,65535,65535,1374,1374,65535,65535,1374,65535,65535,65535,1374,1374,65535,65535,65535,1374,65535,65535,1374],[1374,1374,1374,65535,65535,1374,1374,1374,1374,1374,65535,1374,65535,65535,1374,1374,1374,65535,65535,65535,1374,65535,65535,1374,65535,1374,65535,1374,65535,65535,1374,65535,1374,65535,1374,65535,1374,1374,65535,1374,1374,65535,65535,65535,1374,65535,65535,1374,1374,65535,65535,1374,65535,65535,65535,1374,1374,65535,65535,65535,1374,65535,65535,1374]
```

poblabs, 2022-02-24:
I guess a workaround is to define the animated bitmap as `msg.icon` or something, then use `{{msg.icon}}` in the bitmap animation node. That seems to work better than just using a function to define it

o0shojo0o, 2022-02-26:
Yes you are missing the `animationDelay` parameter! 
https://docs.bastelbunker.de/pixelit/api.html#bitmap-animation

o0shojo0o, 2022-02-26:
```js
msg.bitmapAnimation = {};
msg.bitmapAnimation.animationDelay = 100;
msg.bitmapAnimation.data = "[1374,1374,1374,65535,65535,1374,1374,1374,1374,1374,65535,65535,65535,65535,1374,1374,1374,65535,65535,65535,65535,65535,65535,1374,65535,65535,65535,65535,65535,65535,65535,65535,1374,65535,65535,65535,65535,65535,65535,1374,1374,65535,65535,65535,65535,65535,65535,1374,1374,65535,65535,65535,65535,65535,65535,1374,1374,65535,65535,65535,65535,65535,65535,1374],[1374,1374,1374,65535,65535,1374,1374,1374,1374,1374,65535,65535,65535,65535,1374,1374,1374,65535,65535,65535,65535,65535,65535,1374,65535,65535,65535,65535,65535,65535,65535,65535,1374,65535,65535,65535,65535,65535,65535,1374,1374,65535,65535,65535,65535,65535,65535,1374,1374,65535,65535,65535,65535,65535,65535,1374,1374,65535,65535,65535,1374,65535,65535,1374],[1374,1374,1374,65535,65535,1374,1374,1374,1374,1374,65535,65535,65535,65535,1374,1374,1374,65535,65535,65535,65535,65535,65535,1374,65535,65535,65535,65535,65535,65535,65535,65535,1374,65535,65535,65535,65535,65535,65535,1374,1374,65535,65535,65535,65535,65535,65535,1374,1374,65535,65535,1374,65535,65535,65535,1374,1374,65535,65535,65535,1374,65535,65535,1374],[1374,1374,1374,65535,65535,1374,1374,1374,1374,1374,65535,65535,65535,65535,1374,1374,1374,65535,65535,65535,65535,65535,65535,1374,65535,65535,65535,65535,65535,65535,65535,65535,1374,65535,65535,65535,1374,65535,65535,1374,1374,65535,65535,65535,1374,65535,65535,1374,1374,65535,65535,1374,65535,65535,65535,1374,1374,65535,65535,65535,1374,65535,65535,1374],[1374,1374,1374,65535,65535,1374,1374,1374,1374,1374,65535,65535,65535,65535,1374,1374,1374,65535,65535,65535,65535,65535,65535,1374,65535,65535,65535,1374,65535,65535,65535,65535,1374,65535,65535,65535,1374,1374,65535,1374,1374,65535,65535,65535,1374,65535,65535,1374,1374,65535,65535,1374,65535,65535,65535,1374,1374,65535,65535,65535,1374,65535,65535,1374],[1374,1374,1374,65535,65535,1374,1374,1374,1374,1374,65535,65535,65535,65535,1374,1374,1374,65535,65535,65535,1374,65535,65535,1374,65535,65535,65535,1374,65535,65535,65535,65535,1374,65535,1374,65535,1374,1374,65535,1374,1374,65535,65535,65535,1374,65535,65535,1374,1374,65535,65535,1374,65535,65535,65535,1374,1374,65535,65535,65535,1374,65535,65535,1374],[1374,1374,1374,65535,65535,1374,1374,1374,1374,1374,65535,1374,65535,65535,1374,1374,1374,65535,65535,65535,1374,65535,65535,1374,65535,1374,65535,1374,65535,65535,1374,65535,1374,65535,1374,65535,1374,1374,65535,1374,1374,65535,65535,65535,1374,65535,65535,1374,1374,65535,65535,1374,65535,65535,65535,1374,1374,65535,65535,65535,1374,65535,65535,1374]";
 
return msg;
```

closed 2022-02-26 by o0shojo0o