pixelit-project/PixelIt #147: only 4 leds on

sander816, 2022-02-27:
i am new to pixel it.
I have 8x32 led matrix, trhis one:
https://nl.aliexpress.com/item/33018815831.html?gatewayAdapt=glo2nld&spm=a2g0o.9042311.0.0.27424c4dvlK0EP

i flashed pixelit on a esp12-f. Connected 5v and gnd to led matrix, and d2 on the esp to D-in.
Esp is loading on the ip adress it has, but i only see 4 leds on the matrix that are on.

what am i doing wrong?

o0shojo0o, 2022-03-02:
Which firmware did you use?

o0shojo0o, 2022-03-02:
Yes that's what I was getting at, try the **firmware_ESP8266.bin** from here https://github.com/o0shojo0o/PixelIt/releases

sander816, 2022-03-02:
well, thats what i used already.
I meant regarding hardware

sander816, 2022-03-02:
and just tested the matrix with another controller, they are working fine

o0shojo0o, 2022-03-14:
Yes this could be a problem, if the power supply fluctuates this can even go to a crash

closed 2022-03-14 by o0shojo0o