pixelit-project/PixelIt #172: Can't change wifi or reset

rozza-m, 2022-05-20:
I moved my PixelIt to a different AP (but the same SSID) - doesn't connect.
So I tried to re-associate and re-connect the wifi again - there is no method to do so.
So I tried to factory reset - again - there is no method to do this.

Feels like at least a "reset to defaults" should be available?

rozza-m, 2022-05-21:
N.b. it seems like esp WiFi manager or firmware is pretty basic with WiFi roaming between multiple APs with the same SSID or else this wouldn't be a problem 🙁 https://esp32.com/viewtopic.php?t=3885

o0shojo0o, 2022-05-21:
[image]

closed 2022-05-21 by o0shojo0o

rozza-m, 2022-05-21:
😱 Amazing!

rozza-m, 2022-05-21:
@o0shojo0o that was amazingly fast work, thanks. I'll try it out when the WebUI updates.

o0shojo0o, 2022-05-21:
If you want to test it in advance you can compile the new firmware and start the WebUi via the DevServer. 
https://github.com/o0shojo0o/PixelIt/wiki/Develop-on-WebUI

rozza-m, 2022-05-22:
Thanks 
[image]
Is this the IP of the PixelIt or the dev server?

o0shojo0o, 2022-05-22:
IP of the PixelIt 😊

commit 7c2ff07e (2022-05-21): added 'Wifi Reset' and 'Factory Reset' function

closed #172