pixelit-project/PixelIt #173: Support for HUB75

reptil1990, 2022-05-21:
Hey,

I just found this project and it would be a perfect match for cheap LED Panels I have laying around.
Unfortianatly they are HUB75 and not ws2812b.
Woult it be possible to get support for that?

Here is already a libary ready to go: https://github.com/pixelmatix/SmartMatrix/tree/master

My Matrix has 64x64 and an option to group pixels would also zoom in th pixel art and would not show the too small.

Thank you.

rozza-m, 2022-05-22:
I think if your matrix is not 8x32, this is probably the wrong project for you.

o0shojo0o, 2022-05-22:
Yes sorry @reptil1990 , unfortunately @rozza-m is right, to run this project with a different pixel count everything would have to be reworked.

closed 2022-05-22 by o0shojo0o

reptil1990, 2022-05-23:
Would it not be possible to group Pixels? So that a pixel in the original scale is lighting up 4 pixels in a bigger matrix?
That sould be doable? 
You are the experts just a thought.

foorschtbar, 2022-05-23:
I have a experience with HUB75 Panels and
- they needs at least esp32
- (needs a lot more pins)
- panels are more expensive
- a lot of code must be rewritten

For me it make also no sense. Write your own code to display the PixelIt Stuff on such a display. You need a MQTT Client or REST API, a GFX Libary and you could still use NodeRED and the PixIt Flows . Changing the PixelIt code to support HUB75 it make IHMO absolute NO sense.