pixelit-project/PixelIt #176: Support for Different LED

TravisRoy, 2022-06-27:
However, you can create a custom version where you only need to adjust the code in three places.
https://github.com/o0shojo0o/PixelIt/blob/c3429897945d4815e5970459dc6e1c4cc014d7b5/src/PixelIt.ino#L1990-L2003

For **GRB** it would look like this: 
```cpp
 // Matrix Color Correction 
 if (userLEDCorrection != UncorrectedColor) 
 { 
 	FastLED.addLeds<WS2812B, MATRIX_PIN, GRB>(leds, NUMMATRIX).setCorrection(userLEDCorrection); 
 } 
 else if (userColorTemp != UncorrectedTemperature) 
 { 
 	FastLED.addLeds<WS2812B, MATRIX_PIN, GRB>(leds, NUMMATRIX).setTemperature(userColorTemp); 
 } 
 else 
 { 
 	int *rgbArray = GetUserCutomCorrection(); 
 	FastLED.addLeds<WS2812B, MATRIX_PIN, GRB>(leds, NUMMATRIX).setCorrection(matrix->Color(rgbArray[0], rgbArray[1], rgbArray[2])); 
 } 
```

_Originally posted by @o0shojo0o in https://github.com/o0shojo0o/PixelIt/issues/77#issuecomment-945079691_

TravisRoy, 2022-06-27:
I'm pretty sure I need the WS2812B order.

closed 2023-09-12 by github-actions[bot]

dosipod, 2025-02-19:
The was listed above does not work and will give error .
```
                                                                                       ^
/workspaces/PixelIt/src/PixelIt.ino:3725:87: error: no matching function for call to 'CFastLED::addLeds<NEOPIXEL, 16, RGB>(CRGB [256], int)'
         FastLED.addLeds<NEOPIXEL, MATRIX_PIN , RGB>(leds, MATRIX_WIDTH * MATRIX_HEIGHT).setCorrection(matrix->Color(rgbArray[0], rgbArray[1], rgbArray[2]));
                                                                                       ^
In file included from /workspaces/PixelIt/src/PixelIt.ino:32:
.pio/libdeps/ESP32_generic/FastLED/src/FastLED.h:274:130: note: candidate: 'template<ESPIChipsets CHIPSET, unsigned char DATA_PIN, unsigned char CLOCK_PIN, EOrder RGB_ORDER, unsigned int SPI_DATA_RATE> CLEDController& CFastLED::addLeds(CRGB*, int, int)'
  template<ESPIChipsets CHIPSET,  uint8_t DATA_PIN, uint8_t CLOCK_PIN, EOrder RGB_ORDER, uint32_t SPI_DATA_RATE > CLEDController &addLeds(struct CRGB *data, int nLedsOrOffset, int nLedsIfOffset = 0) {
                                                                                                                                  ^~~~~~~
.pio/libdeps/ESP32_generic/FastLED/src/FastLED.h:274:130: note:   template argument deduction/substitution failed:
.pio/libdeps/ESP32_generic/FastLED/src/FastLED.h:292:95: note: candidate: 'template<ESPIChipsets CHIPSET, unsigned char DATA_PIN, unsigned char CLOCK_PIN> static CLEDController& CFastLED::addLeds(CRGB*, int, int)'
  template<ESPIChipsets CHIPSET,  uint8_t DATA_PIN, uint8_t CLOCK_PIN > static CLEDController &addLeds(struct CRGB *data, int nLedsOrOffset, int nLedsIfOffset = 0) {
                                                                                               ^~~~~~~
.pio/libdeps/ESP32_generic/FastLED/src/FastLED.h:292:95: note:   template argument deduction/substitution failed:
.pio/libdeps/ESP32_generic/FastLED/src/FastLED.h:310:113: note: candidate: 'template<ESPIChipsets CHIPSET, unsigned char DATA_PIN, unsigned char CLOCK_PIN, EOrder RGB_ORDER> static CLEDController& CFastLED::addLeds(CRGB*, int, int)'
  template<ESPIChipsets CHIPSET,  uint8_t DATA_PIN, uint8_t CLOCK_PIN, EOrder RGB_ORDER > static CLEDController &addLeds(struct CRGB *data, int nLedsOrOffset, int nLedsIfOffset = 0) {
                                                                                                                 ^~~~~~~
.pio/libdeps/ESP32_generic/FastLED/src/FastLED.h:310:113: note:   template argument deduction/substitution failed:
.pio/libdeps/ESP32_generic/FastLED/src/FastLED.h:367:25: note: candidate: 'template<template<unsigned char DATA_PIN, EOrder RGB_ORDER> class CHIPSET, unsigned char DATA_PIN, EOrder RGB_ORDER> static CLEDController& CFastLED::addLeds(CRGB*, int, int)'
  static CLEDController &addLeds(struct CRGB *data, int nLedsOrOffset, int nLedsIfOffset = 0) {
                         ^~~~~~~
.pio/libdeps/ESP32_generic/FastLED/src/FastLED.h:367:25: note:   template argument deduction/substitution failed:
.pio/libdeps/ESP32_generic/FastLED/src/FastLED.h:374:25: note: candidate: 'template<template<unsigned char DATA_PIN, EOrder RGB_ORDER> class CHIPSET, unsigned char DATA_PIN> static CLEDController& CFastLED::addLeds(CRGB*, int, int)'
  static CLEDController &addLeds(struct CRGB *data, int nLedsOrOffset, int nLedsIfOffset = 0) {
                         ^~~~~~~
.pio/libdeps/ESP32_generic/FastLED/src/FastLED.h:374:25: note:   template argument deduction/substitution failed:
/workspaces/PixelIt/src/PixelIt.ino:3725:87: error: wrong number of template arguments (3, should be 2)
         FastLED.addLeds<NEOPIXEL, MATRIX_PIN , RGB>(leds, MATRIX_WIDTH * MATRIX_HEIGHT).setCorrection(matrix->Color(rgbArray[0], rgbArray[1], rgbArray[2]));
                                                                                       ^
In file included from /workspaces/PixelIt/src/PixelIt.ino:32:
.pio/libdeps/ESP32_generic/FastLED/src/FastLED.h:381:25: note: candidate: 'template<template<unsigned char DATA_PIN> class CHIPSET, unsigned char DATA_PIN> static CLEDController& CFastLED::addLeds(CRGB*, int, int)'
  static CLEDController &addLeds(struct CRGB *data, int nLedsOrOffset, int nLedsIfOffset = 0) {
                         ^~~~~~~
.pio/libdeps/ESP32_generic/FastLED/src/FastLED.h:381:25: note:   template argument deduction/substitution failed:
/workspaces/PixelIt/src/PixelIt.ino:3725:87: error: wrong number of template arguments (3, should be 2)
         FastLED.addLeds<NEOPIXEL, MATRIX_PIN , RGB>(leds, MATRIX_WIDTH * MATRIX_HEIGHT).setCorrection(matrix->Color(rgbArray[0], rgbArray[1], rgbArray[2]));
                                                                                       ^
In file included from /workspaces/PixelIt/src/PixelIt.ino:32:
.pio/libdeps/ESP32_generic/FastLED/src/FastLED.h:429:25: note: candidate: 'template<template<EOrder RGB_ORDER> class CHIPSET, EOrder RGB_ORDER> static CLEDController& CFastLED::addLeds(CRGB*, int, int)'
  static CLEDController &addLeds(struct CRGB *data, int nLedsOrOffset, int nLedsIfOffset = 0) {
                         ^~~~~~~
.pio/libdeps/ESP32_generic/FastLED/src/FastLED.h:429:25: note:   template argument deduction/substitution failed:
/workspaces/PixelIt/src/PixelIt.ino:3725:87: error: wrong number of template arguments (3, should be 2)
[... cut ...]
        FastLED.addLeds<WS2812B, MATRIX_PIN>(leds, MATRIX_WIDTH * MATRIX_HEIGHT).setCorrection(userLEDCorrection);
    }
    else if (userColorTemp != UncorrectedTemperature)
    {
        FastLED.addLeds<WS2812B, MATRIX_PIN>(leds, MATRIX_WIDTH * MATRIX_HEIGHT).setTemperature(userColorTemp);
    }
    else
    {
        int *rgbArray = GetUserCutomCorrection();
        FastLED.addLeds<WS2812B, MATRIX_PIN>(leds, MATRIX_WIDTH * MATRIX_HEIGHT).setCorrection(matrix->Color(rgbArray[0], rgbArray[1], rgbArray[2]));
    }```