pixelit-project/PixelIt #181: Feature request: allow to remove the small bars below the clock

jghaanstra, 2022-08-08:
Great project, just sponsored it a little.

What is the purpose of the small bars below the clock ? I would like to see a feature to remove them in both the settings and through the API.

o0shojo0o, 2022-08-08:
Hi, 
the bars are the days of the week. 
And if you use the "Big clock font" no weekdays should be displayed anymore.

[image]

jghaanstra, 2022-08-08:
> the bars are the days of the week.

Ah right, makes sense now. I'm aware using the big clock wil remove the bars but I'd still like to see an option to use the small clock without the bars though if it's not too complicated to add.

o0shojo0o, 2022-08-08:
I will have a look at it the days, I think that it does not need so much effort.

o0shojo0o, 2022-08-16:
available in version 1.1.0

closed 2022-08-16 by o0shojo0o