pixelit-project/PixelIt #186: No output on firmware 1.1.0

jghaanstra, 2022-08-16:
I just updated to firmware 1.1.0 but it's not working as expected. There is not output shown on the LED matrix other than a couple of dots which are visible upon reboot. WebUI is working fine, REST API is working fine, MQTT is working fine but nothing is shown.

I tried a factory reset to no avail. Even tried reverting the firmware to 1.0.0 but that hasnt worked either.

o0shojo0o, 2022-08-16:
My two PixelIt show no errors, you could try the [CleanFlash.bin](https://pixelit-project.github.io/firmware.html#flashing).

jghaanstra, 2022-08-16:
Found the issue, my Wemos D1 Mini is having identity issues. When I first connected the matrix I had to connect it to pin D5 (which was different to the documentation). With the update I had to revert back to pin D2 to make the LED matrix work. Strange.

closed 2022-08-16 by jghaanstra

o0shojo0o, 2022-08-16:
This is really strange, but no matter main thing :)