pixelit-project/PixelIt #195: Rotated tiled 4x 8x8 CJMCU if upgrade firmware

Giuseppe84, 2022-09-05:
Hi everyone. thanks for you Pixelit project. I am using it with four tiled 8x8 CJMCU matrix, and the latest  firmware version  I can properly use is  0.3.20. If I upgrade the firmware and set "type 3 - tiled 4x 8x8 CJMCU" as the matrix type, each matrix appears rotated 90 °, I have tried other "matrix type" settings but none works well. The only solution could be to rotate the matrix, but I can't solve the problem in this way, they are glued on the support.

foorschtbar, 2022-09-05:
Thats fascinating. My four tiled 8x8 CJMCU matrix doesn't work since i fixed it in 0.4. I think we have connected the tiles differently. Could you send me a picture of the back so I can see how the tiles are connected? I suspect that your pin headers are arranged verticall - this results in a continuous zigzag arrangement across the entire matrix. On my matrix, the pin headers are horizontally, the result is, that the zigzag must jumps after each tile.

tl;dr: send me a picture for confirmation and that we could add a second "8x8 CJMCU matrix" type which work as before.

(DIN is in the top right corner in the image)
[image]

Giuseppe84, 2022-09-05:
Of course, I hope the image is clear, unfortunately I put a lot of glue. DIN is in the bottom right corner of each matrix
[image]

foorschtbar, 2022-09-05:
Please test this version. Select Type 3.

[firmware_v1.2.0_d1_mini.zip](https://github.com/pixelit-project/PixelIt/files/9489819/firmware_v1.2.0_d1_mini.zip)

Giuseppe84, 2022-09-05:
Works perfectly! Thank you very much

PR #196 (2022-09-05): Readded support for tiled 4x 8x8 CJMCU Matrix in column major order
Fixes #195