pixelit-project/PixelIt #199: MQTT Buttons state not updated - Home Assistant

fmunozs, 2022-09-07:
Hello, I've just installed the latest version and connected it with my local Home Assistant server.

I can see correctly the LED Matrix state, Illuminance sensor, however none of the 3 buttons are showing up its state.  The state is being shown on the PixelIt panel correctly but not on MQTT.

[image]

o0shojo0o, 2022-09-08:
This might be related to the change in version [1.0.0 (2022-07-22)](https://github.com/pixelit-project/PixelIt#100-2022-07-22),  
but I don't really know about Home Assistant... @foorschtbar or @pplucky  can have a look over here?

meli-femunoz, 2022-09-08:
Just want do add that I also tried the 0.0.4 version and the events didnt show up either on Home Assistant.

fmunozs, 2022-09-09:
hello pplucky! thanks for pointing me to MQTT Explorer. I can see the message going to the sever, however it doesnt seem to be using the "automatic" format that Home Assistant uses and that must be why its not detected. I will see if I can fix it. thanks

pplucky, 2022-09-09:
> hello pplucky! thanks for pointing me to MQTT Explorer. I can see the message going to the sever, however it doesnt seem to be using the "automatic" format that Home Assistant uses and that must be why its not detected. I will see if I can fix it. thanks

I can also confirm that this is pre [1.0.0](https://github.com/pixelit-project/PixelIt#100-2022-07-22) behavior, as I didn't touch the button code (I had no way of testing it at the time).

As per my analysis, it seems the MQTT discovery topic is being sent from PixelIT to HA, but without any value published. If you look [here](https://github.com/pixelit-project/PixelIt/blob/50fca73264f167576079150faabf315be87f865a/src/PixelIt.ino#L2324),  you'll notice that unlike what happens for other sensors, no line [like this one](https://github.com/pixelit-project/PixelIt/blob/50fca73264f167576079150faabf315be87f865a/src/PixelIt.ino#L2381), exists for the buttons.

Maybe @o0shojo0o can help figure out where are the values to be published (the same as in the UI). so that this can be fixed.

o0shojo0o, 2022-09-10:
> Maybe @o0shojo0o can help figure out where are the values to be published (the same as in the UI). so that this can be fixed.

In teorie, the event `<mqttMasterTopic>/buttons/ ` should come one of these JSON `{"leftButton":true}` , `{"middleButton":true}` or `{"rightButton":true}`, or just with false.
Here is the line: https://github.com/pixelit-project/PixelIt/blob/25faa6663117982816e3cc5e9d25993425500913/src/PixelIt.ino#L883

pplucky, 2022-09-11:
@o0shojo0o @fmunozs @meli-femunoz Is any of you willing to try my [fork](https://github.com/pplucky/PixelIt)?

Using @o0shojo0o's tip above, I managed to fix a couple of things that didn't seem OK to me and renamed the buttons to Left, Middle and Right instead of just Button0, Button1 and Button2 (which still stands as the sensor ID).

Playing around with MQTT, I managed to get these to appear:
[image]

But as I don't have buttons assembled, I cannot do the real testing.

If it looks OK, I can create a PR for it.

fmunozs, 2022-09-11:
Hello pplucky, I tried your fork but still cant see the value on HA.  I see the update events on the "buttons" topic, and see the new val_tpl  but it doesnt show yet. Also, I believe that the current implementation of MQTT for button state can be improved because currently its only possible to see the state of a single button. If I press 2 or 3 buttons at the same time only the last press is reflected.

[image]

I will try to send a PR in an hour, I also think that it is better to change the dev_cla for the sensors because it is not a timestamp.

pplucky, 2022-09-11:
> Hello pplucky, I tried your fork but still cant see the value on HA. I see the update events on the "buttons" topic, and see the new val_tpl but it doesnt show yet. Also, I believe that the current [...]

Yes, that's a fact. My only purpose at this time was to make it MQTT discovery work also for buttons.

Actually, there is a single MQTT topic for all buttons which is being overwritten each time a button is pressed, but this has nothing to do with MQTT discovery, as discovery is only getting information from the existing MQTT topics to feed the discovery-generated entities.

> [image]
> 
> I will try to send a PR in an hour, I also think that it is better to change the dev_cla for the sensors because it is not a timestamp.

Damn, I forgot to remove the dev_cla, that is why it is not working. Fixed now in my [fork](https://github.com/pplucky/PixelIt).

fmunozs, 2022-09-12:
Hello! the last one shows the value now, great! 

[image]

pplucky, 2022-09-12:
> Hello! the last one shows the value now, great! I am checking your fork to try and send a new PR to allow multiple buttons.
> 
> [image]

Untested, but tried creating a new MQTT topic under buttons named after the corresponding button (button1, button2 or button3) and use its state to feed MQTT discovery directly...

fmunozs, 2022-09-12:
Tried to compile it but it fails. 

```
C:/Users/fmunozs/Desktop/fork/src/PixelIt.ino: In function 'boolean MQTTreconnect()':
C:/Users/fmunozs/Desktop/fork/src/PixelIt.ino:2459:89: error: conversion from 'const __FlashStringHelper*' to 'const StringSumHelper' is ambiguous
                 payload.replace(F("#STATETOPIC#"), String(F("buttons/button") + String(n)));
                                                                                         ^
```

pplucky, 2022-09-12:
> Tried to compile it but it fails.
>
> ```
> C:/Users/fmunozs/Desktop/fork/src/PixelIt.ino: In function 'boolean MQTTreconnect()':
> C:/Users/fmunozs/Desktop/fork/src/PixelIt.ino:2459:89: error: conversion [...]

Can you please try again?

pplucky, 2022-09-12:
> Works great! Pressed the middle and right button and both show as pressed.

Thanks for letting me know. I think i'd rather transform these into separate binary sensors before submitting a PR. Tomorrow is another day.

fmunozs, 2022-09-12:
Everything seems to be working OK! I had to remove the device from HA because the old button sensors configuration appeared next to the new ones, maybe its a good idea to mention that somewhere for new users.

https://user-images.githubusercontent.com/1041604/189560776-d3b326ed-8948-4152-afcc-1001be5ac5f6.mp4

PR #209 (2022-09-14): Fix MQTT discovery for buttons (now binary sensors with separate MQTT topics)
Fix MQTT discovery for buttons and change entity names for buttons, from Button0/1/2 to Left/Middle/Right Button.

Fixes #199.

Also changed the MQTT Discovery button entities from sensors to binary_sensors (with icon). This is a breaking change that causes the old non-working sensors to be kept and new working binary sensors to also appear. Removing the PixelIt device from MQTT Discovery integration should do the trick, as it will appear again automatically with only thee correct sensors..

Also allowed all button a [... cut]
closed 2022-09-14 by o0shojo0o