pixelit-project/PixelIt #226: ESP32 Fixes

TravisRoy, 2022-10-23:
Unable to save config fix:

`    // Mounting FileSystem
    Serial.println(F("Mounting file system..."));
#if defined(ESP8266)
    if (LittleFS.begin())
#elif defined(ESP32)
    #define FORMAT_SPIFFS_IF_FAILED true
    if (SPIFFS.begin(FORMAT_SPIFFS_IF_FAILED))
    
#endif`

Wifi AP fix:

`  // Hostname
    if (hostname.isEmpty())
    {
        hostname = "PixelIt";
        
    }
    WiFi.hostname(hostname);

    WiFi.mode(WIFI_AP);

    wifiManager.setAPCallback(EnteredHotspotCallback);
    wifiManager.setMinimumSignalQuality();
    // Timout for the wifi connection until the hotspot is set up`

Bettman66, 2022-10-25:
Thank you, it works.

o0shojo0o, 2022-12-26:
@rliegmann can you evaluate if this fix is still necessary?

rliegmann, 2022-12-27:
@TravisRoy FORMAT_SPIFFS_IF_FAILED is useful. It can happen that the memory is not initialized. I'd say we'll take over.

Why do you think we need WiFi.mode(WIFI_AP)? I just tried it with a new ESP32 and it builds the AP cleanly. What problems did you have?

DSRocker, 2023-01-01:
I tried the new release (firmware_v2.2-beta-ONEWIRE_wemos_d1_mini32.bin) with a new ESP32 Dev Kit C V4. 
No AP

DSRocker, 2023-01-07:
I get everytime this error in serial monitor: (no AP)
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57

rliegmann, 2023-01-10:
Have you tried completely clearing the memory of the ESP?
Then flashed again?

DSRocker, 2023-01-11:
Yes, I did! And I tried another esp32.. the same errors

closed 2023-09-12 by github-actions[bot]

commputethis, 2023-12-09:
I get the same thing.  I have tried on 3 different ESP32 device models (multiple boards of each model) and get the same error @DSRocker was getting.

foorschtbar, 2023-12-10:
> I get the same thing.  I have tried on 3 different ESP32 device models (multiple boards of each model) and get the same error @DSRocker was getting.

Please try the a full upgrade from the webflasher (se documentation) and select the erase option. We have many working esp32, that must be a filesystem error related to missing/wrong formatting

commputethis, 2023-12-10:
I've done that and no change. I've tried from an Ubuntu machine and a Windows machine. I've tried from two different networks too (two different Internet providers also). One of the networks had no filtering to make sure something wasn't getting blocked.

I've also tried downloading and flashing. Esp8266 devices work without issue. It's only esp32 that I can't get to work.

foorschtbar, 2023-12-10:
Could you post the serial log after flashing with webflasher?

commputethis, 2023-12-10:
Well, I had found something online about SPIFFS and found for many that flashing it with the SPIFFS_Test.ino and then flashing with the actual config wanted fixes it.  Now, I am not getting that error, but one about unknown pins for D3 and D1.  I have attached the logs from a generic esp32 board and a wemo d1 mini esp32 board.  Both have the same errors.  Probably need to do a new problem since it is now a different error.
[esp-web-tools-logs-wemos-d1-mini-esp32.txt](https://github.com/pixelit-project/PixelIt/files/13629247/esp-web-tools-logs-wemos-d1-mini-esp32.txt)
[esp-web-tools-logs-generic-esp32.txt](https://github.com/pixelit-project/PixelIt/files/13629248/esp-web-tools-logs-generic-esp32.txt)

commputethis, 2023-12-10:
Looking at the boards, neither one has a D1 or D3 pin.
[image]
[image]

commputethis, 2023-12-10:
I guess it could be the ```[BH1750] ERROR: other error``` that is causing the issue.