pixelit-project/PixelIt #229: The matrix keeps dark

davidlamd89, 2022-11-10:
I'm trying to get this to work, I've followed the instructions, but the 8x32 matrix keeps dark, I tried with different boards (Wemos D1 mini clones), I also tried to feed the Matrix only with USB or with an external power source, but I still can't get it to work.
The matrix lights up when using WLED, so i believe it's a configuration problem.

Can someone help me? 

Thanks in advance

o0shojo0o, 2022-11-10:
Did you use the [firmware_v2.1.2_d1_mini.bin](https://github.com/pixelit-project/PixelIt/releases/download/2.1.2/firmware_v2.1.2_d1_mini.bin)?

davidlamd89, 2022-11-10:
> Did you use the [firmware_v2.1.2_d1_mini.bin](https://github.com/pixelit-project/PixelIt/releases/download/2.1.2/firmware_v2.1.2_d1_mini.bin)?

Yes I did.
 I've solved the problem, I had to use pin D4 instead of D2 as described in Wiki. 

Thank you for taking the time on trying to help.

closed 2022-11-10 by o0shojo0o