pixelit-project/PixelIt #246: ESP8266-01S

DSRocker, 2022-12-27:
Hi,
after flashing to my ESP-01S I can't see any AP...
It tested it with:
firmware_v2.1.2_d1_mini.bin
firmware_v2.1.2_ESP8266.bin
firmware_v2.2-beta_d1_mini.bin
firmware_v2.2-beta_ESP8266.bin
& I compiled it on myself...
=> no AP

rliegmann, 2022-12-27:
Hi, can you show us what comes over the serial interface?
LG

DSRocker, 2022-12-27:
Retrieving maximum program size .pio\build\ESP8266\firmware_v2.2-beta-ONEWIRE_ESP8266.elf
Checking size .pio\build\ESP8266\firmware_v2.2-beta-ONEWIRE_ESP8266.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=====     ]  53.4% (used 43716 bytes from 81920 bytes)
Flash: [=====     ]  46.2% (used 482056 bytes from 1044464 bytes)
Configuring upload protocol...
AVAILABLE: espota, esptool
CURRENT: upload_protocol = esptool
Looking for upload port...
Auto-detected: COM7
Uploading .pio\build\ESP8266\firmware_v2.2-beta-ONEWIRE_ESP8266.bin
esptool.py v3.0
Serial port COM7
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 40:91:51:4f:77:1a
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Compressed 486208 bytes to 339033...
Writing at 0x00000000... (4 %)
Writing at 0x00004000... (9 %)
Writing at 0x00008000... (14 %)
Writing at 0x0000c000... (19 %)
Writing at 0x00010000... (23 %)
Writing at 0x00014000... (28 %)
Writing at 0x00018000... (33 %)
Writing at 0x0001c000... (38 %)
Writing at 0x00020000... (42 %)
Writing at 0x00024000... (47 %)
Writing at 0x00028000... (52 %)
Writing at 0x0002c000... (57 %)
Writing at 0x00030000... (61 %)
Writing at 0x00034000... (66 %)
Writing at 0x00038000... (71 %)
Writing at 0x0003c000... (76 %)
Writing at 0x00040000... (80 %)
Writing at 0x00044000... (85 %)
Writing at 0x00048000... (90 %)
Writing at 0x0004c000... (95 %)
Writing at 0x00050000... (100 %)
Wrote 486208 bytes (339033 compressed) at 0x00000000 in 36.2 seconds (effective 107.5 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
=================================================================== [SUCCESS] Took 56.67 seconds ===================================================================
Processing d1_mini (platform: espressif8266@2.6.3; board: d1_mini; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------Ve

rliegmann, 2022-12-27:
Okay, and what comes out of the serial interface after the upload?

DSRocker, 2022-12-27:
14:33:07.161 -> ⸮⸮⸮⸮rl⸮$⸮c|⸮⸮⸮⸮{⸮clc⸮⸮o'⸮lno⸮⸮⸮bx⸮l{l{$⸮s⸮⸮n⸮c⸮ocp

rliegmann, 2022-12-27:
Okay because the baud rate is not correct.
Is this set to 115200 in PlatformIO (monitor_speed)?

DSRocker, 2022-12-27:
--- Terminal on COM7 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
{$␀l��|␀�d�|␃␌␄␌�␌l�␌#|��␃�␛�s�bd␌c��'o�$g'���␄c␜p��dsdrd␛�{�o�#�o#x␀�

rliegmann, 2022-12-27:
Mhh now I don't know what to do, because unfortunately I don't have an ESP-8266-01 with me.
This might help: [Link](https://www.kollino.de/arduino/esp8266-01-wlan-modul-mit-dem-arduino-uno-programmieren/)
Here is something about the fact that RX and TX must not be connected.
Do example programs work accordingly?

DSRocker, 2022-12-27:
Is it possible to try it on an ESP32?

rliegmann, 2022-12-27:
In any case.  :-)
I use an ESP32. To be precise, an ESP32-WROOM-32.
The firmware wemos_d1_mini32 works here.

DSRocker, 2022-12-27:
OK. Thx. I tried with ESP8266-F; it works. What is the IP to access after connecting to the AP?

DSRocker, 2022-12-27:
I've got it ;-)
Thanks a lot... I would like to try it with some 8x8 WS2812 & 16x16 WS2812. I hope it'll work fine for me.

DSRocker, 2022-12-27:
my Matrix doesn't show anything....

DSRocker, 2022-12-27:
---- Opened the serial port COM7 ----
{$l��|�d�|�l�#|����s�#dc��'o�do'���cp�dsdrl8�o��$��b'�|�l��#��og�l��l �'od`gs��o#$`;��gcl`�c��$r�`��n�Mounting file system...
Mounted file system.
[1970-00-00T00:00:00] LoadConfig: Loaded
[BH1750] ERROR: received NACK on transmit of address
[1970-00-00T00:00:00] Setup: BMP280 Trying
[1970-00-00T00:00:00] Setup: No BMP280, BME280, BME 680 or DHT Sensor found
Framebuffer_GFX::begin Width: 32 Height: 8 Num Pixels: 256
[1970-00-00T00:00:00] Setup: Software Serial started
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Status:
*WM: 3
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 3
*WM: IP Address:
*WM: 192.168.178.4
[1970-01-01T00:00:06] Setup: Wifi connected...yeey :)
[1970-01-01T00:00:06] Setup: Local IP
[1970-01-01T00:00:06] Setup: 1xx.xxx.xxx.xxx
[1970-01-01T00:00:06] Setup: 1xx.xxx.xxx.xxx
[1970-01-01T00:00:06] Setup: 2xx.xxx.xxx.xxx
[1970-01-01T00:00:06] Setup: Starting UDP
[1970-01-01T00:00:06] Setup: Webserver started
[1970-01-01T00:00:06] DFPlayer: DFPlayer started
[1970-01-01T00:00:06] Sync TimeServer: 1xx.xxx.xxx.xxx waiting for sync
[2022-12-27T17:57:00] WebSocketEvent: [0] Connected from 1xx.xxx.xxx.xxx url: /
[2022-12-27T17:57:23] WebSocketEvent: Incoming JSON (Length: 109)
[2022-12-27T17:57:23] CreateFrames: JSON contains Text (Length: 95)
[2022-12-27T17:57:24] CheckUpdate: Checking...
[2022-12-27T17:57:25] CheckUpdate: New FW available
[2022-12-27T17:57:33] WebSocketEvent: Incoming JSON (Length: 175)
[2022-12-27T17:57:33] CreateFrames: JSON contains Text (Length: 161)

rliegmann, 2022-12-31:
What kind of LED matrix do you have?
Which one is selected in WebConfig?
How is it wired?

LG

DSRocker, 2022-12-31:
This one: [WS2812 Matrix](https://www.az-delivery.de/products/rgb-led-panel-ws2812b-16x16-256-leds-flexibel-led-modul-5050smd-ic-einzeladressierbare-vollfarbfunktionen-mit-dc5v-kompatibel-mit-raspberry-pi) 
I tried all options in the webConfig and restarted every time after changing.

The matrix works with WLED flashed; same wiring as with PixelIt.

foorschtbar, 2022-12-31:
For this matrix size you must change the code. Out of the box is only 8x32 Pixel supported. if the matrix keeps dark, you use the wrong pin? Have you checked the documentation?

DSRocker, 2022-12-31:
What Do I needle to change ?
I used GPIO 4 (D2 on D1 mini)?! Correct ?

DSRocker, 2022-12-31:
It's IO2 on ESP8266. I need to change the code for my matrix. Could you please help me how to do it.
Thanks a lot

closed 2023-01-01 by DSRocker