pixelit-project/PixelIt #248: VSC Compiling ends with errors - don't know why

DSRocker, 2023-01-01:
here some issues:
In file included from .pio\libdeps\ESP8266\SHA-1 Hash\src\Hash.cpp:28:0:
.pio\libdeps\ESP8266\SHA-1 Hash\src\sha1/sha1.h:19:0: warning: "ICACHE_FLASH_ATTR" redefined [enabled by default]
 #define ICACHE_FLASH_ATTR   __attribute__((section(".irom0.text")))
 ^
In file included from C:\...\.platformio\packages\framework-arduinoespressif8266@3.20704.0\cores\esp8266/esp8266_peri.h:24:0,
                 from C:\...\.platformio\packages\framework-arduinoespressif8266@3.20704.0\cores\esp8266/Arduino.h:38,
                 from .pio\libdeps\ESP8266\SHA-1 Hash\src\Hash.cpp:23:
C:\...\.platformio\packages\framework-arduinoespressif8266@3.20704.0\tools\sdk\include/c_types.h:89:0: note: this is the location of the previous definition   
 #define ICACHE_FLASH_ATTR   __attribute__((section("\".irom0.text." __FILE__ "." __ICACHE_STRINGIZE(__LINE__) "." __ICACHE_STRINGIZE(__COUNTER__) "\"")))

-----------------
In file included from .pio\libdeps\ESP8266\FastLED\src/FastLED.h:67:0,
                 from D:/PixelIt/PixelIt/src/PixelIt.ino:31:
.pio\libdeps\ESP8266\FastLED\src/fastspi.h:145:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"

--------------
Archiving .pio\build\ESP8266\lib851\libWebSockets.a
.pio\libdeps\ESP8266\PubSubClient\src\PubSubClient.cpp: In member function 'boolean PubSubClient::publish_P(const char*, const uint8_t*, unsigned int, boolean)':
.pio\libdeps\ESP8266\PubSubClient\src\PubSubClient.cpp:523:19: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
     return (rc == expectedLength);

-----------
.pio\libdeps\ESP8266\ColorConverter\src\ColorConverterLib.cpp: In static member function 'static void ColorConverter::RgbToHsl(uint8_t, uint8_t, uint8_t, double&, double&, double&)':
.pio\libdeps\ESP8266\ColorConverter\src\ColorConverterLib.cpp:75:9: warning: 'h' may be used uninitialized in this function [-Wmaybe-uninitialized]
   h /= 6;
         ^
.pio\libdeps\ESP8266\ColorConverter\src\ColorConverterLib.cpp: In static member function 'static void ColorConverter::HsvToRgb(double, double, double, uint8_t&, uint8_t&, uint8_t&)':
.pio\libdeps\ESP8266\ColorConverter\src\ColorConverterLib.cpp:111:32: warning: 'b' may be used uninitialized in this function [-Wmaybe-uninitialized]
  blue = static_cast<uint8_t>(b * 255);
                                ^
.pio\libdeps\ESP8266\ColorConverter\src\ColorConverterLib.cpp:110:33: warning: 'g' may be used uninitialized in this function [-Wmaybe-uninitialized]
  green = static_cast<uint8_t>(g * 255);
                                 ^
.pio\libdeps\ESP8266\ColorConverter\src\ColorConverterLib.cpp:109:31: warning: 'r' may be used uninitialized in this function [-Wmaybe-uninitialized]
  red = static_cast<uint8_t>(r * 255);

--------------
In file included from .pio\libdeps\ESP8266\FastLED\src/FastLED.h:67:0,
                 from .pio\libdeps\ESP8266\Framebuffer GFX/Framebuffer_GFX.h:33,
                 from .pio\libdeps\ESP8266\FastLED NeoMatrix/FastLED_NeoMatrix.h:23,
                 from .pio\libdeps\ESP8266\FastLED NeoMatrix\FastLED_NeoMatrix.cpp:38:
.pio\libdeps\ESP8266\FastLED\src/fastspi.h:145:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
                       ^
In file included from .pio\libdeps\ESP8266\FastLED\src/FastLED.h:67:0,
                 from .pio\libdeps\ESP8266\Framebuffer GFX/Framebuffer_GFX.h:33,
                 from .pio\libdeps\ESP8266\Framebuffer GFX\Framebuffer_GFX.cpp:38:
.pio\libdeps\ESP8266\FastLED\src/fastspi.h:145:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"

------------------
  File "C:\...\.platformio\packages\framework-arduinoespressif8266@3.20704.0\tools\elf2bin.py", line 182, in <module>
    sys.exit(main())
  File "C:\...\.platformio\packages\framework-arduinoespressif8266@3.20704.0\tools\elf2bin.py", line 163, in main
    wrapper(
  File "C:\...\.platformio\packages\framework-arduinoespressif8266@3.20704.0\tools\elf2bin.py", line 161, in wrapper
    write_bin(out=out, args=args, **kwargs)
  File "C:\...\.platformio\packages\framework-arduinoespressif8266@3.20704.0\tools\elf2bin.py", line 68, in write_bin
    entry = int(get_elf_entry( elf, args.path ))
  File "C:\...\.platformio\packages\framework-arduinoespressif8266@3.20704.0\tools\elf2bin.py", line 44, in get_elf_entry
    raise Exception('Unable to find entry point in file "' + elf + '"')
Exception: Unable to find entry point in file "C:\...\.platformio\packages\framework-arduinoespressif8266@3.20704.0\bootloaders\eboot\eboot.elf"
*** [.pio\build\ESP8266\firmware_v2.2-beta-ONEWIRE_ESP8266.bin] Error 1

=================================================================== [FAILED] Took 34.43 seconds ===================================================================

Environment      Status    Duration
---------------  --------  ------------
wemos_d1_mini32  FAILED    00:00:39.944
ESP8266          FAILED    00:00:34.363
d1_mini          FAILED    00:00:36.352
nodemcuv2        FAILED    00:00:34.430
============================================================== 4 failed, 0 succeeded in 00:02:25.089 ==============================================================

o0shojo0o, 2023-01-02:
I can't confirm that, it runs through on me and on git.
https://github.com/pixelit-project/PixelIt/actions/runs/3779771521/jobs/6425316227#step:9:1896

DSRocker, 2023-01-02:
Ok. Thx. I'll delete all the packages and try again.
But only next week.
Another question: is there any guides to run this great PixelIt :-) with 16x16 WS2812 MATRIX?
THANKS

o0shojo0o, 2023-01-02:
No, there is no such thing, it is not supported out of the box.

aptonline, 2023-02-06:
I believe the error is just a warning and can be disregarded. I had the same errors and from my research its just to let you know that hardware SPI pins aren't supported with FastLED.