pixelit-project/PixelIt #260: error in lof on flash

bobdivx, 2023-02-12:
Hello, on esp32

[02:51:35][1970-01-01T00:00:39] SendTelemetry: Sending...
[02:51:35][1970-01-01T00:00:39] Sync TimeServer: de.pool.ntp.org waiting for sync
[02:51:35][2023-02-12T02:51:36] Auto Brightness: Lux: 0.00 set brightness to 20
[02:51:35][2023-02-12T02:51:36] WebSocketEvent: [0] Connected from 10.1.0.142 url: /
[02:51:35][ 40070][E][vfs_api.cpp:24] open(): File system is not mounted
[02:51:35][ 40075][E][vfs_api.cpp:24] open(): File system is not mounted
[02:51:35][ 40075][E][vfs_api.cpp:24] open(): File system is not mounted
[02:51:35][ 40076][E][vfs_api.cpp:24] open(): File system is not mounted
[02:51:35][ 40082][E][vfs_api.cpp:24] open(): File system is not mounted
[02:51:35][ 40087][E][vfs_api.cpp:24] open(): File system is not mounted
[02:51:35][ 40093][E][vfs_api.cpp:24] open(): File system is not mounted
[02:51:35][ 40098][E][vfs_api.cpp:24] open(): File system is not mounted
[02:51:35][ 40104][E][vfs_api.cpp:24] open(): File system is not mounted
[02:51:35][ 40109][E][vfs_api.cpp:24] open(): File system is not mounted
[02:51:35][2023-02-12T02:51:36] WebSocketEvent: [1] Connected from 10.1.0.142 url: /
[02:51:35][ 40236][E][vfs_api.cpp:24] open(): File system is not mounted
[02:51:35][ 40242][E][vfs_api.cpp:24] open(): File system is not mounted
[02:51:35][ 40246][E][vfs_api.cpp:24] open(): File system is not mounted
[02:51:35][ 40247][E][vfs_api.cpp:24] open(): File system is not mounted
[02:51:35][ 40248][E][vfs_api.cpp:24] open(): File system is not mounted
[02:51:35][ 40254][E][vfs_api.cpp:24] open(): File system is not mounted
[02:51:35][ 40259][E][vfs_api.cpp:24] open(): File system is not mounted
[02:51:35][ 40265][E][vfs_api.cpp:24] open(): File system is not mounted
[02:51:35][ 40271][E][vfs_api.cpp:24] open(): File system is not mounted
[02:51:35][ 40276][E][vfs_api.cpp:24] open(): File system is not mounted

aptonline, 2023-02-12:
This is usually caused by there being no file system partition on the esp32 so there is no where for the config to be saved. I installed from source and added a `data` folder to the root of the code folder.

I’m not sure the pixelate firmware.bin creates this at the same time when flashing.

rliegmann, 2023-03-30:
Hi,
Have a look here: #129 
Maybe that is a solution. Let me know if it snapped. Otherwise we can close the issue.
LG Ralf

closed 2023-09-12 by github-actions[bot]