pixelit-project/PixelIt #285: bme680 Reading Backwards?

Keshin1988, 2023-05-22:
hey all im unsure if this is my scenario but i feel like the voc reading on my bme680 is reading in reverse?
according to the index  sheet it shows
[image]

my findings are outside in fresh clean air im getting approx 200-210kohm if i directly breath onto it, trying to exhale (increase CO2) the level drops considerably to 100kohms 

according to the index above shouldn't my value be increasing rather than decreasing when i breath on it

Keshin1988, 2023-05-22:
It's 6am in the morning and it's been left outside in the clean air all night and this is the results I am getting now. 
[image]

foorschtbar, 2023-05-23:
I have no BME680 try test it, but you can upload a clean example sketch and test only the BME680.

closed 2023-09-12 by github-actions[bot]