pixelit-project/PixelIt #287: SAVE CONFIG button not clickable ESP32 d1mini esp32-d0wdq5 rev 3

DSRocker, 2023-06-18:
Hi,
I use d1mini for siveral projects. I flashed the newest 2.2.0 ([firmware_v2.2.0_wemos_d1_mini32.bin](https://github.com/pixelit-project/PixelIt/releases/download/2.2.0/firmware_v2.2.0_wemos_d1_mini32.bin))
It works fine with nodered; but I can't save the configs over the webUI.
The button is not clickable. After configuration pixelIT doesn't save the configs.
What to do?
Thx

foorschtbar, 2023-06-19:
Please try a private browser tab, delete cache and disable add-ons, they try it again. there is no known issue with the web ui

DSRocker, 2023-06-19:
I tried it with Chrome and Safari.
see attachment (Screenshots, Fotos)
IP: ...235: ESP8266
IP: ...234: ESP32: "SAVE CONFIG" not possible
[https://drive.google.com/drive/folders/1W878EsqUMwiDXferPK8ozZJezjIms3Ja?usp=sharing](url)

foorschtbar, 2023-06-19:
The ESP32 WebUI doenst load settings. Save is only possible, if there anything you can save. Please show me the browser console log.

DSRocker, 2023-06-19:
[2023-06-19T11:52:14] CheckUpdate: No new FW available
[2023-06-19T11:52:13] CheckUpdate: Checking...
[2023-06-19T11:52:13] SendTelemetry: Sending...
[2023-06-19T11:52:10] WebSocketEvent: [0] Disconnected!
[2023-06-19T11:52:01] WebSocketEvent: [2] Connected from 192.168.178.10 url: /
[2023-06-19T11:51:49] WebSocketEvent: [1] Connected from 192.168.178.10 url: /

DSRocker, 2023-06-19:
after activating NodeRedFlow:
[2023-06-19T11:53:21] CreateFrames: JSON contains Text (Length: 199)
[2023-06-19T11:53:21] HandleScreen: Incoming JSON length: 199
[2023-06-19T11:53:18] CreateFrames: JSON contains Text (Length: 199)
[2023-06-19T11:53:18] HandleScreen: Incoming JSON length: 199
[2023-06-19T11:53:14] CreateFrames: JSON contains Text (Length: 196)
[2023-06-19T11:53:14] HandleScreen: Incoming JSON length: 196
[2023-06-19T11:53:12] CreateFrames: JSON contains Text (Length: 199)
[2023-06-19T11:53:11] HandleScreen: Incoming JSON length: 199
[2023-06-19T11:53:09] CreateFrames: JSON contains Text (Length: 199)
[2023-06-19T11:53:09] HandleScreen: Incoming JSON length: 199
[2023-06-19T11:52:14] CheckUpdate: No new FW available
[2023-06-19T11:52:13] CheckUpdate: Checking...
[2023-06-19T11:52:13] SendTelemetry: Sending...
[2023-06-19T11:52:10] WebSocketEvent: [0] Disconnected!
[2023-06-19T11:52:01] WebSocketEvent: [2] Connected from 192.168.178.10 url: /
[2023-06-19T11:51:49] WebSocketEvent: [1] Connected from 192.168.178.10 url: /

DSRocker, 2023-06-19:
oh... ;-)
what should i look for?
=> screenshot of an issue: [https://drive.google.com/file/d/1wrFQHLPl--SESFctqUjPpgnfLyWJGLpo/view?usp=drive_link](url)

DSRocker, 2023-06-19:
<html>
<body>
<div tabindex="-1" class="console-message-wrapper console-error-level" style="box-sizing: border-box; min-width: 0px; min-height: 0px; display: flex; border-top-width: ; border-top-style: ; border-top-color: var(--override-error-border-color); border-bottom: 1px solid transparent; --console-color-black: #000; --console-color-red: #a00; --console-color-green: #0a0; --console-color-yellow: #a50; --console-color-blue: #00a; --console-color-magenta: #a0a; --console-color-cyan: #0aa; --console-color-gray: #aaa; --console-color-darkgray: #555; --console-color-lightred: #f55; --console-color-lightgreen: #5f5; --console-color-lightyellow: #ff5; --console-color-lightblue: #55f; --console-color-ightmagenta: #f5f; --console-color-lightcyan: #5ff; --console-color-white: #fff; background-color: var(--override-console-error-background-color); color: rgb(32, 33, 36); font-family: consolas, &quot;lucida console&quot;, &quot;courier new&quot;, monospace; font-size: 12px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; white-space: normal; text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;"><div class="console-message" style="box-sizing: border-box; min-width: 0px; min-height: 17px; clear: right; position: relative; padding: 3px 22px 1px 0px; margin-left: 24px; flex: 1 1 auto; display: flex;"><span class="source-code" style="box-sizing: border-box; min-width: 0px; min-height: 0px; font-family: var(--source-code-font-family); white-space: pre-wrap; font-size: var(--source-code-font-size) !important; flex: 1 1 auto; line-height: 1.2;"><span class="console-message-anchor" style="box-sizing: border-box; min-width: 0px; min-height: 0px; float: right; text-align: right; max-width: 100%; margin-left: 4px;"><span class="devtools-link" role="link" tabindex="-1" style="box-sizing: border-box; min-width: 0px; min-height: 0px; color: var(--color-text-secondary); text-decoration: underline; outline-offset: 2px; cursor: pointer; word-break: break-all;">/#/:1</span> </span><span class="console-message-text" style="box-sizing: border-box; min-width: 0px; min-height: 0px; color: var(--override-error-text-color) !important;">Uncaught <span class="object-value-error source-code" style="box-sizing: border-box; min-width: 0px; min-height: 0px; font-family: var(--source-code-font-family); white-space: pre-wrap; font-size: var(--source-code-font-size) !important; line-height: 1.2;"><span style="box-sizing: border-box; min-width: 0px; min-height: 0px;">SyntaxError: Unexpected token '}', "{"config":}" is not valid JSON
    at JSON.parse (&lt;anonymous&gt;)
<span style="box-sizing: border-box; min-width: 0px; min-height: 0px;">    at t.value (<span class="devtools-link" role="link" tabindex="-1" title="webpack:///node_modules/vue-native-websocket/dist/build.js:1:6015" style="box-sizing: border-box; min-width: 0px; min-height: 0px; color: var(--color-text-secondary); text-decoration: underline; outline-offset: 2px; cursor: pointer; word-break: break-all;">build.js:1:6015</span>)
</span><span style="box-sizing: border-box; min-width: 0px; min-height: 0px;">    at t.value (<span class="devtools-link" role="link" tabindex="-1" title="webpack:///node_modules/vue-native-websocket/dist/build.js:1:5840" style="box-sizing: border-box; min-width: 0px; min-height: 0px; color: var(--color-text-secondary); text-decoration: underline; outline-offset: 2px; cursor: pointer; word-break: break-all;">build.js:1:5840</span>)
</span><span style="box-sizing: border-box; min-width: 0px; min-height: 0px;">    at forEach.t.WebSocket.&lt;computed&gt; (<span class="devtools-link" role="link" tabindex="-1" title="webpack:///node_modules/vue-native-websocket/dist/build.js:1:5540" style="box-sizing: border-box; min-width: 0px; min-height: 0px; color: var(--color-text-secondary); text-decoration: underline; outline-offset: 2px; cursor: pointer; word-break: break-all;">build.js:1:5540</span>)</span></span></span></span></span></div></div><div tabindex="-1" class="console-message-wrapper console-error-level" style="box-sizing: border-box; min-width: 0px; min-height: 0px; display: flex; border-top-width: ; border-top-style: ; border-top-color: var(--override-error-border-color); border-bottom: 1px solid transparent; --console-color-black: #000; --console-color-red: #a00; --console-color-green: #0a0; --console-color-yellow: #a50; --console-color-blue: #00a; --console-color-magenta: #a0a; --console-color-cyan: #0aa; --console-color-gray: #aaa; --console-color-darkgray: #555; --console-color-lightred: #f55; --console-color-lightgreen: #5f5; --console-color-lightyellow: #ff5; --console-color-lightblue: #55f; --console-color-ightmagenta: #f5f; --console-color-lightcyan: #5ff; --console-color-white: #fff; background-color: var(--override-console-error-background-color); color: rgb(32, 33, 36); font-family: consolas, &quot;lucida console&quot;, &quot;courier new&quot;, monospace; font-size: 12px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; white-space: normal; text-decoration-thickness: initial; text-decoration [... cut ...]
  | i | @ | vue-router.esm.js:2008
  | i | @ | vue-router.esm.js:2008
  | Nt | @ | vue-router.esm.js:2012
  | te.confirmTransition | @ | vue-router.esm.js:2392
  | te.transitionTo | @ | vue-router.esm.js:2260
  | e.push | @ | vue-router.esm.js:2715
  | _e.push | @ | vue-router.esm.js:3037
  | _ | @ | vue-router.esm.js:1139
  | Ge | @ | vue.runtime.esm.js:2988
  | n | @ | vue.runtime.esm.js:1786
  | Co.o._wrapper | @ | vue.runtime.esm.js:7405

</span></div></div></div></div></div>
</body>
</html>

foorschtbar, 2023-06-19:
There is a problem with the config file. please erase the complete flash: https://randomnerdtutorials.com/esp32-erase-flash-memory/

DSRocker, 2023-06-19:
Is it ok to do this with "ESP32 Download Tool" V 3.9.2?
What do I have to consider while flashing? SPI-Mode: QIO or DIO / QOUT or DOUT?!

DSRocker, 2023-06-19:
I'll try a new ESP32 d1_mini :-)

DSRocker, 2023-06-19:
I tried with a new one... the same... 
in serial monitoring:
[ 51412][E][vfs_api.cpp:24] open(): File system is not mounted

foorschtbar, 2023-06-19:
Maybe @rliegmann can help, he fixed the ESP32 support. My PixelIts runs fine with ESP8266

DSRocker, 2023-06-19:
OK. It will be great. Yes, on ESP8266 it works fine...

rliegmann, 2023-06-25:
Hi foorschtbar,

i can try to help you What board do you have exactly?
LG Ralf

DSRocker, 2023-06-25:
Esp32 d1 mini (MH-ET).

rliegmann, 2023-06-25:
OK, Try to lash the following code. It tests the file system. Then again the Pixelit am over it.
```
#include "FS.h"
#include "SPIFFS.h"

/* You only need to format SPIFFS the first time you run a
   test or else use the SPIFFS plugin to create a partition
   https://github.com/me-no-dev/arduino-esp32fs-plugin */
#define FORMAT_SPIFFS_IF_FAILED true

void listDir(fs::FS &fs, const char * dirname, uint8_t levels){
    Serial.printf("Listing directory: %s\r\n", dirname);

    File root = fs.open(dirname);
    if(!root){
        Serial.println("- failed to open directory");
        return;
    }
    if(!root.isDirectory()){
        Serial.println(" - not a directory");
        return;
    }

    File file = root.openNextFile();
    while(file){
        if(file.isDirectory()){
            Serial.print("  DIR : ");
            Serial.println(file.name());
            if(levels){
                listDir(fs, file.path(), levels -1);
            }
        } else {
            Serial.print("  FILE: ");
            Serial.print(file.name());
            Serial.print("\tSIZE: ");
            Serial.println(file.size());
        }
        file = root.openNextFile();
    }
}

void readFile(fs::FS &fs, const char * path){
    Serial.printf("Reading file: %s\r\n", path);

    File file = fs.open(path);
    if(!file || file.isDirectory()){
        Serial.println("- failed to open file for reading");
        return;
    }

    Serial.println("- read from file:");
    while(file.available()){
        Serial.write(file.read());
    }
    file.close();
}

void writeFile(fs::FS &fs, const char * path, const char * message){
    Serial.printf("Writing file: %s\r\n", path);

    File file = fs.open(path, FILE_WRITE);
    if(!file){
        Serial.println("- failed to open file for writing");
        return;
    }
    if(file.print(message)){
        Serial.println("- file written");
    } else {
        Serial.println("- write failed");
    }
    file.close();
}

void appendFile(fs::FS &fs, const char * path, const char * message){
    Serial.printf("Appending to file: %s\r\n", path);

    File file = fs.open(path, FILE_APPEND);
    if(!file){
        Serial.println("- failed to open file for appending");
        return;
    }
    if(file.print(message)){
        Serial.println("- message appended");
    } else {
        Serial.println("- append failed");
    }
    file.close();
}

void renameFile(fs::FS &fs, const char * path1, const char * path2){
    Serial.printf("Renaming file %s to %s\r\n", path1, path2);
    if (fs.rename(path1, path2)) {
        Serial.println("- file renamed");
    } else {
        Serial.println("- rename failed");
    }
}

void deleteFile(fs::FS &fs, const char * path){
    Serial.printf("Deleting file: %s\r\n", path);
    if(fs.remove(path)){
        Serial.println("- file deleted");
    } else {
        Serial.println("- delete failed");
    }
}

void testFileIO(fs::FS &fs, const char * path){
    Serial.printf("Testing file I/O with %s\r\n", path);

    static uint8_t buf[512];
    size_t len = 0;
    File file = fs.open(path, FILE_WRITE);
    if(!file){
        Serial.println("- failed to open file for writing");
        return;
    }

    size_t i;
    Serial.print("- writing" );
    uint32_t start = millis();
    for(i=0; i<2048; i++){
        if ((i & 0x001F) == 0x001F){
          Serial.print(".");
        }
        file.write(buf, 512);
    }
    Serial.println("");
    uint32_t end = millis() - start;
    Serial.printf(" - %u bytes written in %u ms\r\n", 2048 * 512, end);
    file.close();

    file = fs.open(path);
    start = millis();
    end = start;
    i = 0;
    if(file && !file.isDirectory()){
        len = file.size();
        size_t flen = len;
        start = millis();
        Serial.print("- reading" );
        while(len){
            size_t toRead = len;
            if(toRead > 512){
                toRead = 512;
            }
            file.read(buf, toRead);
            if ((i++ & 0x001F) == 0x001F){
              Serial.print(".");
            }
            len -= toRead;
        }
        Serial.println("");
        end = millis() - start;
        Serial.printf("- %u bytes read in %u ms\r\n", flen, end);
        file.close();
    } else {
        Serial.println("- failed to open file for reading");
    }
}

void setup(){
    Serial.begin(115200);
    if(!SPIFFS.begin(FORMAT_SPIFFS_IF_FAILED)){
        Serial.println("SPIFFS Mount Failed");
        return;
    }
    
    listDir(SPIFFS, "/", 0);
    writeFile(SPIFFS, "/hello.txt", "Hello ");
    appendFile(SPIFFS, "/hello.txt", "World!\r\n");
    readFile(SPIFFS, "/hello.txt");
    renameFile(SPIFFS, "/hello.txt", "/foo.txt");
    readFile(SPIFFS, "/foo.txt");
    deleteFile(SPIFFS, "/foo.txt");
    testFileIO(SPIFFS, "/test.txt");
    deleteFile(SPIFFS, "/test.txt");
    Serial.println( "Test complete" );
}

void loop(){

}
```

DSRocker, 2023-08-02:
how to do it?

DSRocker, 2023-08-03:
after flashing:
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
invalid header: 0x29294346
invalid header: 0x29294346
invalid header: 0x29294346
invalid header: 0x29294346
invalid header: 0x29294346
invalid header: 0x29294346
invalid header: 0x29294346
invalid header: 0x29294346

DSRocker, 2023-08-03:
Using 'COM7' as serial port.
Connecting......
Detecting chip type... ESP32
Connecting....

Chip Info:
 - Chip Family: ESP32
 - Chip Model: ESP32-D0WDQ5-V3 (revision 3)
 - Number of Cores: 2
 - Max CPU Frequency: 240MHz
 - Has Bluetooth: YES
 - Has Embedded Flash: NO
 - Has Factory-Calibrated ADC: YES
 - MAC Address: E0:5A:1B:A0:1A:98
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
 - Flash Size: 4MB
 - Flash Mode: dio
 - Flash Frequency: 80MHz
Erasing flash (this may take a while)...
Chip erase completed successfully in 13.6s
Flash will be erased from 0x00001000 to 0x00005fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x00058fff...
Compressed 17120 bytes to 11164...
Writing at 0x00001000... (100 %)
Wrote 17120 bytes (11164 compressed) at 0x00001000 in 0.5 seconds (effective 297.8 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 129...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (129 compressed) at 0x00008000 in 0.0 seconds (effective 573.1 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 811.3 kbit/s)...
Hash of data verified.
Compressed 297584 bytes to 170400...
Writing at 0x00010000... (9 %)
Writing at 0x0001b7c8... (18 %)
Writing at 0x00024791... (27 %)
Writing at 0x00029c80... (36 %)
Writing at 0x0002f0e3... (45 %)
Writing at 0x000341e9... (54 %)
Writing at 0x0003987c... (63 %)
Writing at 0x00041e3a... (72 %)
Writing at 0x0004ba6a... (81 %)
Writing at 0x00050e81... (90 %)
Writing at 0x00056737... (100 %)
Wrote 297584 bytes (170400 compressed) at 0x00010000 in 4.0 seconds (effective 595.4 kbit/s)...
Hash of data verified.

Leaving...
Hard Resetting...
Hard resetting via RTS pin...
Done! Flashing is complete!

Showing logs:
[07:44:20]E (141) esp_core_dump_flashrcore dump partition found!
[07:44:20]E (141) esp_core_dump_flash: No core dump partition found!
[07:44:20]E (10) SPIFFS: mount failed, -10025
[07:44:25]Listing directory: /
[07:44:25]Writing file: /hello.txt
[07:44:25]- file written
[07:44:25]Appending to file: /hello.txt
[07:44:25]- message appended
[07:44:25]Reading file: /hello.txt
[07:44:25]- read from file:
[07:44:25]Hello World!
[07:44:25]Renaming file /hello.txt to /foo.txt
[07:44:25]- file renamed
[07:44:25]Reading file: /foo.txt
[07:44:25]- read from file:
[07:44:25]Hello World!
[07:44:25]Deleting file: /foo.txt
[07:44:25]- file deleted
[07:44:25]Testing file I/O with /test.txt

DSRocker, 2023-08-03:
flashing pixelit.bin V2.2.0

Using 'COM7' as serial port.
Connecting......
Detecting chip type... ESP32
Connecting....

Chip Info:
 - Chip Family: ESP32
 - Chip Model: ESP32-D0WDQ5-V3 (revision 3)
 - Number of Cores: 2
 - Max CPU Frequency: 240MHz
 - Has Bluetooth: YES
 - Has Embedded Flash: NO
 - Has Factory-Calibrated ADC: YES
 - MAC Address: E0:5A:1B:A0:1A:98
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
 - Flash Size: 4MB
 - Flash Mode: dio
 - Flash Frequency: 40MHz
Erasing flash (this may take a while)...
Chip erase completed successfully in 13.3s
Flash will be erased from 0x00001000 to 0x00005fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x00119fff...
Compressed 17104 bytes to 11191...
Writing at 0x00001000... (100 %)
Wrote 17104 bytes (11191 compressed) at 0x00001000 in 0.5 seconds (effective 301.5 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 129...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (129 compressed) at 0x00008000 in 0.0 seconds (effective 601.0 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 821.4 kbit/s)...
Hash of data verified.
Compressed 1088448 bytes to 676531...
Writing at 0x00010000... (2 %)
Writing at 0x0001a3d2... (4 %)
Writing at 0x00026d9e... (7 %)
Writing at 0x000304a3... (9 %)
Writing at 0x0003b9d4... (11 %)
Writing at 0x00044cf4... (14 %)
Writing at 0x0004bbfb... (16 %)
Writing at 0x000526d2... (19 %)
Writing at 0x00057e09... (21 %)
Writing at 0x0005d572... (23 %)
Writing at 0x00064073... (26 %)
Writing at 0x0006b196... (28 %)
Writing at 0x00070a0c... (30 %)
Writing at 0x00076852... (33 %)
Writing at 0x0007c0cc... (35 %)
Writing at 0x00081eb5... (38 %)
Writing at 0x00087398... (40 %)
Writing at 0x0008c6a9... (42 %)
Writing at 0x00091915... (45 %)
Writing at 0x00096c7b... (47 %)
Writing at 0x0009beda... (50 %)
Writing at 0x000a113c... (52 %)
Writing at 0x000a63e8... (54 %)
Writing at 0x000ab6a2... (57 %)
Writing at 0x000b0bd1... (59 %)
Writing at 0x000b6847... (61 %)
Writing at 0x000bbd9b... (64 %)
Writing at 0x000c0fd4... (66 %)
Writing at 0x000c63de... (69 %)
Writing at 0x000cb939... (71 %)
Writing at 0x000d0f78... (73 %)
Writing at 0x000d6732... (76 %)
Writing at 0x000dc4d3... (78 %)
Writing at 0x000e1e63... (80 %)
Writing at 0x000e7942... (83 %)
Writing at 0x000f02de... (85 %)
Writing at 0x000f88c5... (88 %)
Writing at 0x000fd91d... (90 %)
Writing at 0x00107c33... (92 %)
Writing at 0x0010d15b... (95 %)
Writing at 0x0011268a... (97 %)
Writing at 0x00117fcb... (100 %)
Wrote 1088448 bytes (676531 compressed) at 0x00010000 in 15.7 seconds (effective 555.8 kbit/s)...
Hash of data verified.

Leaving...
Hard Resetting...
Hard resetting via RTS pin...
Done! Flashing is complete!

Showing logs:
[07:47:37]E (445) esp_core_dump_flash: No core dump partition found!
[07:47:37]E (445) esp_core_dump_flash: No core dump partition found!
[07:47:37]Mounting file system...
[07:47:37]E (22) SPIFFS: mount failed, -10025
[07:47:37][    31][E][SPIFFS.cpp:89] begin(): Mounting SPIFFS failed! Error: -1
[07:47:37]Failed to mount FS
[07:47:37][BH1750] ERROR: received NACK on transmit of address
[07:47:37][1970-00-00T00:00:00] Setup: BMP280 Trying
[07:47:37][1970-00-00T00:00:00] Setup: No BMP280, BME280 or BME 680 sensor found
[07:47:38][1970-00-00T00:00:00] Setup: No DHT Sensor found
[07:47:38]Framebuffer_GFX::begin Width: 32 Height: 8 Num Pixels: 256
[07:47:38][1970-00-00T00:00:00] Setup: Software Serial started
[07:47:40]*wm:AutoConnect 
[07:47:41]*wm:No wifi saved, skipping 
[07:47:41]*wm:AutoConnect: FAILED 
[07:47:41]*wm:StartAP with SSID:  PIXELIT
[07:47:41]*wm:AP IP address: 192.168.4.1
[07:47:41][1970-01-01T00:00:03] Hotspot: Waiting for WiFi configuration
[07:47:42]*wm:Starting Web Portal

rliegmann, 2023-08-06:
I have now tried it with various ESP32 boards.
Has always worked. I suspect that this is a specific problem with this board.
The way I read it, you have now taken a different ESP.
Does that work for you then? If so, we can close the Issus.

LG

DSRocker, 2023-08-06:
I tried it with the same Version of the Board; don't work :-/
I don't know why.
If you like you can close this issue. I have to find the reason for this issue by my self.

Thanks

closed 2023-08-06 by DSRocker