pixelit-project/PixelIt #312: Is that correct? "Pin-Zuordnung: Unbekannter Pin"

nox309, 2023-10-08:
I am currently testing the new beta 2.4.0 with the Ulanzi.
Works great so far it also works all buttons.
I noticed in the log when booting, however, that he announces an unknown pin assignment. 

Is that correct and are the pins that are not further defined, or is that an "error" in the pre-assignmentof the pins for the Ulanzi?

Log: 
[2023-10-08T22:31:33] SaveConfig: Saved
Mounting file system...
Mounted file system.
[1970-00-00T00:00:00] LoadConfig: Loaded
[BH1750] ERROR: received NACK on transmit of address
[1970-00-00T00:00:00] Setup: SHT31 Trying
[1970-00-00T00:00:00] Setup: SHT31 started
Framebuffer_GFX::begin Width: 32 Height: 8 Num Pixels: 256
[1970-00-00T00:00:00] Pin Assignment: Unknown Pin
[1970-00-00T00:00:00] Pin Assignment: Unknown Pin
[1970-00-00T00:00:00] Setup: Software Serial started
parsed json
*wm:AutoConnect
*wm:Connecting to SAVED AP: x.x
*wm:connectTimeout not set, ESP waitForConnectResult...
*wm:AutoConnect: SUCCESS
*wm:STA IP Address: 10.10.x.x
[1970-01-01T00:00:04] Setup: Wifi connected...yeey :)
[1970-01-01T00:00:04] Setup: Local IP
[1970-01-01T00:00:04] Setup: 10.10.x.x
[1970-01-01T00:00:04] Setup: 10.10.x.x
[1970-01-01T00:00:04] Setup: 255.255.255.0
[1970-01-01T00:00:04] Setup: Starting UDP
[1970-01-01T00:00:04] Setup: Web server started
[1970-01-01T00:00:04] DFPlayer: DFPlayer started

Further, I noticed that during the boot process 100% charge is displayed, but in the dashboard 83%. How are the gradations for the display here?

o0shojo0o, 2023-10-13:
#316

jekader, 2023-10-13:
This is quite easy to reproduce without a physical device. Just upload a binary built against the ulanzi target to this Wokwi sketch:

https://wokwi.com/projects/378474114936175617

It won't be able to mount the FS but will display the log messages and initial bootup screen

jekader, 2023-10-23:
After upgrading my Ulanzi to 2.4.3 I see the log full of these messages:

```
[2023-10-24T00:21:14] Pin assignment - unknown pin: Pin_D6
[2023-10-24T00:21:14] Pin assignment - unknown pin: Pin_D5
[2023-10-24T00:21:14] Pin assignment - unknown pin: Pin_D4
```

foorschtbar, 2023-10-28:
Fixed with #316

closed 2023-10-28 by foorschtbar