pixelit-project/PixelIt #314: Question/Discussion: Reduce MQTT "noise"

oxivanisher, 2023-10-11:
This is not a problem but more a question or discussion.

I realized that the clock sends about once a second a update to MQTT which generates some "noise". In my opinion it would easily be enough if this happens every X seconds or so. This could just be a configuration field for the MQTT config with a seconds value in it.
Also since I have a LDR (analog) on my clock, it would be enough IMHO to send this value every X seconds.

foorschtbar, 2023-10-11:
Do you have an example of the these "noise" messages?

oxivanisher, 2023-10-11:
Sure:
```bash
$ time mqtt sub -T --topic 'pixelit/#'
pixelit/hangar-clock/state: connected
pixelit/hangar-clock/luxsensor: {"lux":0.01}
pixelit/hangar-clock/matrixinfo: {"pixelitVersion":"2.3.2","note":"Hangar Clock","hostname":"hangar-clock","freeSketchSpace":2654208,"wifiRSSI":-61,"wifiQuality":78,"wifiSSID":"thunderiot","ipAddress":"10.5.22.195","freeHeap":17776,"currentMatrixBrightness":1,"wifiBSSID":"F6:9F:C2:34:E5:92","sketchSize":487840,"chipID":11686568,"cpuFreqMHz":80,"sleepMode":false,"uptime":28587,"resetReason":"Software/System restart","matrixsize":{"cols":32,"rows":8}}
pixelit/hangar-clock/dhtsensor: {"humidity":"Not installed","temperature":"Not installed","pressure":"Not installed","gas":"Not installed"}
pixelit/hangar-clock/sensor: {"humidity":"Not installed","temperature":"Not installed","pressure":"Not installed","gas":"Not installed"}
pixelit/hangar-clock/matrixinfo: {"pixelitVersion":"2.3.2","note":"Hangar Clock","hostname":"hangar-clock","freeSketchSpace":2654208,"wifiRSSI":-61,"wifiQuality":78,"wifiSSID":"thunderiot","ipAddress":"10.5.22.195","freeHeap":17776,"currentMatrixBrightness":1,"wifiBSSID":"F6:9F:C2:34:E5:92","sketchSize":487840,"chipID":11686568,"cpuFreqMHz":80,"sleepMode":false,"uptime":28590,"resetReason":"Software/System restart","matrixsize":{"cols":32,"rows":8}}
pixelit/hangar-clock/matrixinfo: {"pixelitVersion":"2.3.2","note":"Hangar Clock","hostname":"hangar-clock","freeSketchSpace":2654208,"wifiRSSI":-61,"wifiQuality":78,"wifiSSID":"thunderiot","ipAddress":"10.5.22.195","freeHeap":17776,"currentMatrixBrightness":1,"wifiBSSID":"F6:9F:C2:34:E5:92","sketchSize":487840,"chipID":11686568,"cpuFreqMHz":80,"sleepMode":false,"uptime":28593,"resetReason":"Software/System restart","matrixsize":{"cols":32,"rows":8}}
pixelit/hangar-clock/matrixinfo: {"pixelitVersion":"2.3.2","note":"Hangar Clock","hostname":"hangar-clock","freeSketchSpace":2654208,"wifiRSSI":-61,"wifiQuality":78,"wifiSSID":"thunderiot","ipAddress":"10.5.22.195","freeHeap":17776,"currentMatrixBrightness":1,"wifiBSSID":"F6:9F:C2:34:E5:92","sketchSize":487840,"chipID":11686568,"cpuFreqMHz":80,"sleepMode":false,"uptime":28596,"resetReason":"Software/System restart","matrixsize":{"cols":32,"rows":8}}
pixelit/hangar-clock/matrixinfo: {"pixelitVersion":"2.3.2","note":"Hangar Clock","hostname":"hangar-clock","freeSketchSpace":2654208,"wifiRSSI":-61,"wifiQuality":78,"wifiSSID":"thunderiot","ipAddress":"10.5.22.195","freeHeap":17776,"currentMatrixBrightness":1,"wifiBSSID":"F6:9F:C2:34:E5:92","sketchSize":487840,"chipID":11686568,"cpuFreqMHz":80,"sleepMode":false,"uptime":28599,"resetReason":"Software/System restart","matrixsize":{"cols":32,"rows":8}}
pixelit/hangar-clock/matrixinfo: {"pixelitVersion":"2.3.2","note":"Hangar Clock","hostname":"hangar-clock","freeSketchSpace":2654208,"wifiRSSI":-61,"wifiQuality":78,"wifiSSID":"thunderiot","ipAddress":"10.5.22.195","freeHeap":17776,"currentMatrixBrightness":1,"wifiBSSID":"F6:9F:C2:34:E5:92","sketchSize":487840,"chipID":11686568,"cpuFreqMHz":80,"sleepMode":false,"uptime":28602,"resetReason":"Software/System restart","matrixsize":{"cols":32,"rows":8}}
pixelit/hangar-clock/matrixinfo: {"pixelitVersion":"2.3.2","note":"Hangar Clock","hostname":"hangar-clock","freeSketchSpace":2654208,"wifiRSSI":-61,"wifiQuality":78,"wifiSSID":"thunderiot","ipAddress":"10.5.22.195","freeHeap":17776,"currentMatrixBrightness":1,"wifiBSSID":"F6:9F:C2:34:E5:92","sketchSize":487840,"chipID":11686568,"cpuFreqMHz":80,"sleepMode":false,"uptime":28605,"resetReason":"Software/System restart","matrixsize":{"cols":32,"rows":8}}
pixelit/hangar-clock/matrixinfo: {"pixelitVersion":"2.3.2","note":"Hangar Clock","hostname":"hangar-clock","freeSketchSpace":2654208,"wifiRSSI":-61,"wifiQuality":78,"wifiSSID":"thunderiot","ipAddress":"10.5.22.195","freeHeap":17776,"currentMatrixBrightness":1,"wifiBSSID":"F6:9F:C2:34:E5:92","sketchSize":487840,"chipID":11686568,"cpuFreqMHz":80,"sleepMode":false,"uptime":28608,"resetReason":"Software/System restart","matrixsize":{"cols":32,"rows":8}}
pixelit/hangar-clock/matrixinfo: {"pixelitVersion":"2.3.2","note":"Hangar Clock","hostname":"hangar-clock","freeSketchSpace":2654208,"wifiRSSI":-64,"wifiQuality":72,"wifiSSID":"thunderiot","ipAddress":"10.5.22.195","freeHeap":17776,"currentMatrixBrightness":1,"wifiBSSID":"F6:9F:C2:34:E5:92","sketchSize":487840,"chipID":11686568,"cpuFreqMHz":80,"sleepMode":false,"uptime":28611,"resetReason":"Software/System restart","matrixsize":{"cols":32,"rows":8}}
pixelit/hangar-clock/luxsensor: {"lux":0.009}
pixelit/hangar-clock/matrixinfo: {"pixelitVersion":"2.3.2","note":"Hangar Clock","hostname":"hangar-clock","freeSketchSpace":2654208,"wifiRSSI":-61,"wifiQuality":78,"wifiSSID":"thunderiot","ipAddress":"10.5.22.195","freeHeap":17776,"currentMatrixBrightness":1,"wifiBSSID":"F6:9F:C2:34:E5:92","sketchSize":487840,"chipID":11686568,"cpuFreqMHz":80,"sleepMode":false,"uptime":28614,"resetReason":"Software/System restart","matrixsize":{"cols":32,"rows":8}}
pixelit/hangar-clock/luxsensor: {"lux":0.01}
pixelit/hangar-clock/matrixinfo: {"pixelitVersion":"2.3.2","note":"Hangar Clock","hostname":"hangar-clock","freeSketchSpace":2654208,"wifiRSSI":-61,"wifiQuality":78,"wifiSSID":"thunderiot","ipAddress":"10.5.22.195","freeHeap":17776,"currentMatrixBrightness":1,"wifiBSSID":"F6:9F:C2:34:E5:92","sketchSize":487840,"chipID":11686568,"cpuFreqMHz":80,"sleepMode":false,"uptime":28617,"resetReason":"Software/System restart","matrixsize":{"cols":32,"rows":8}}
[... cut ...] fiQuality":76,"wifiSSID":"thunderiot","ipAddress":"10.5.22.195","freeHeap":17776,"currentMatrixBrightness":1,"wifiBSSID":"F6:9F:C2:34:E5:92","sketchSize":487840,"chipID":11686568,"cpuFreqMHz":80,"sleepMode":false,"uptime":28620,"resetReason":"Software/System restart","matrixsize":{"cols":32,"rows":8}}
^C
real    0m35.479s
user    0m7.800s
sys     0m0.621s
```
All this was within 35 seconds including the connection time of the java based mqtt client. Its the `matrixinfo` and `luxsensor` messages.

foorschtbar, 2023-10-11:
Thats too mutch. Let me fix that in the next release. Thank u for the hint.

oxivanisher, 2023-10-11:
Thank you for the fast reply! :)

foorschtbar, 2023-10-13:
Interval is now 10 seconds. Change is on develop branch and will be on released in the next two weeks.

commit 9a6523b4 (2023-10-13): Optimized interval of sending mqtt/websocket messages (fixes #314)
closed 2023-10-13 by foorschtbar