pixelit-project/PixelIt #318: Black screen after update new firmware. 2.4.3

StonikMCZ, 2023-10-23:
Hi, after the update was released I uploaded it via the web interface and the LEDs don't light up, if I upload the older FW the LEDs light up again and it works without any problems. Was there any change made in the connection for the new FW? This is a migration from version 2.3.2 to version 2.4.3.

Thank you for the reply.

jekader, 2023-10-23:
Hi, do you see anything at all? I'm also experiencing some display issues on Ulanzi - I get the boot logo, then the zZz and no time display although the preview window suggests it should be displayed.

StonikMCZ, 2023-10-24:
I don't have any buttons or sensors.
When I connect to the web interface I can see what should be visible on the display but the LEDs are not lit.

foorschtbar, 2023-10-24:
Which board do you use? Is you matrix on D2?

pevau, 2023-10-24:
I've got the same problem.
After the upgrade and every start / reboot I get zZZ and then a black screen.
matrixinfo via MQTT says "sleepMode" : true -> black screen.
I can't wake it up.

Board: Wemos D1 Mini
Firmware: firmware_v2.4.3_ESP8266_d1_mini.bin
DIN: D2

foorschtbar, 2023-10-24:
Are you using the sleepmode?

Please try factory reset. The sleepscreen is new, but we had many users without problem. i cant catch it at the moment...

davidlamd89, 2023-10-25:
I'm also having this problem. Alreaddy tryed to do factory reset bu the proble persists.
I downgraded to the previous version and everything is working fine.

Any help?

Thanks

foorschtbar, 2023-10-25:
I will try to reproduce it at the weekend. 

You guys use the generic esp8266 version?

davidlamd89, 2023-10-26:
I've tried both the generic and d1 mini version for ESP8266. Also I'm not using sleep mode.

foorschtbar, 2023-10-28:
Could u guys share you config here? Please update to 2.42 again und visit `http://<matrix>/api/config` and past it here

pevau, 2023-10-28:
Update to 2.4.2 or to 2.4.3?

foorschtbar, 2023-10-28:
> Update to 2.4.2 or to 2.4.3?

Sry, typo. FW 2.4.3.

foorschtbar, 2023-10-28:
And `/api/matrixinfo`

pevau, 2023-10-29:
firmware_v2.4.3_ESP8266_d1_mini.bin

/api/config
{"version":"2.4.3","isESP8266":true,"temperatureUnit":0,"matrixBrightnessAutomatic":false,"mbaDimMin":20,"mbaDimMax":100,"mbaLuxMin":0,"mbaLuxMax":400,"matrixBrightness":20,"matrixType":1,"note":"","hostname":"PixelIt","matrixTempCorrection":"default","ntpServer":"de.pool.ntp.org","clockTimeZone":1,"clockColor":"#FFFFFF","clockSwitchAktiv":true,"clockSwitchSec":6,"clock24Hours":true,"clockDayLightSaving":true,"clockWithSeconds":false,"clockAutoFallbackActive":false,"clockAutoFallbackTime":30,"clockAutoFallbackAnimation":1,"clockDateDayMonth":true,"clockDayOfWeekFirstMonday":true,"clockBlinkAnimated":true,"clockFatFont":false,"clockDrawWeekDays":true,"scrollTextDefaultDelay":100,"bootScreenAktiv":false,"bootBatteryScreen":false,"bootSound":false,"mqttAktiv":true,"mqttUser":"***","mqttPassword":"***","mqttServer":"***","mqttMasterTopic":"Haus/PixelIt/","mqttPort":1883,"luxOffset":0,"temperatureOffset":0,"humidityOffset":0,"pressureOffset":0,"gasOffset":0,"dfpRXpin":"Pin_D7","dfpTXpin":"Pin_D8","onewirePin":"Pin_D1","SCLPin":"Pin_D1","SDAPin":"Pin_D3","btn0Pin":"Pin_D0","btn0PressedLevel":0,"btn0Enabled":false,"btn0Action":2,"btn1Pin":"Pin_D4","btn1PressedLevel":0,"btn1Enabled":false,"btn1Action":1,"btn2Pin":"Pin_D5","btn2PressedLevel":0,"btn2Enabled":false,"btn2Action":0,"ldrDevice":"GL5516","ldrPulldown":10000,"ldrSmoothing":0,"initialVolume":10,"sendTelemetry":true,"checkUpdateScreen":true}

/api/matrixinfo
{"pixelitVersion":"2.4.3","note":"","hostname":"PixelIt","buildSection":"ESP8266_d1_mini","freeSketchSpace":2650112,"wifiRSSI":-46,"wifiQuality":100,"wifiSSID":"***","ipAddress":"***","freeHeap":16904,"currentMatrixBrightness":20,"wifiBSSID":"***","sketchSize":495552,"chipID":4033016,"cpuFreqMHz":80,"sleepMode":true,"uptime":103,"resetReason":"Software/System restart","matrixsize":{"cols":32,"rows":8}}

foorschtbar, 2023-10-30:
@pevau which backend do you use? Node-Red, HA...?

pevau, 2023-10-30:
> @pevau which backend do you use? Node-Red, HA...?

thanks for formatting my comment :-)
I am using Node-Red to control my PixelIt and fill it with content.
The only use case  where I directly control it from HA is controlling the Sleep Mode.

foorschtbar, 2023-10-30:
Ah okay. Here is another issue #323, also with HA and the sleepmode. I will try to reproduce it in the next days.

foorschtbar, 2023-11-04:
Hey guys, you all have Hassio running?

foorschtbar, 2023-11-06:
Please test the [2.4.6-beta](https://github.com/pixelit-project/PixelIt/releases/tag/2.4.6-beta) release and give us feedback

foorschtbar, 2023-11-18:
Please test the [2.4.8-beta](https://github.com/pixelit-project/PixelIt/releases/tag/2.4.8-beta) release and give us feedback.

davidlamd89, 2024-01-05:
I apologize for not responding, I've tried the latest 2.5.0 release, but the problem still appears.
I downgraded to version 2.3.2 and everything is working fine.

foorschtbar, 2024-01-05:
> I apologize for not responding, I've tried the latest 2.5.0 release, but the problem still appears. I downgraded to version 2.3.2 and everything is working fine.

Do you see the boot animation?

davidlamd89, 2024-01-05:
I installed the nodemcuv2 version and it seems to be working.