pixelit-project/PixelIt #319: Feature request:  config backup/restore

jekader, 2023-10-23:
Would be nice to have a feature to dump/restore the config - this makes upgrades safer and also allows to clone settings across different devices.  It can be done in the web UI directly to also lint the config before applying to avoid bricking the device.

foorschtbar, 2023-10-24:
Good idea. API for this should already be there, would just be some work in the WebUI. Maybe @o0shojo0o want to have a look at it?

o0shojo0o, 2023-11-04:
is available in the current beta https://github.com/pixelit-project/PixelIt/releases/tag/2.4.5-beta

jekader, 2023-11-05:
Thank you, updated to 2.4.5 and I can confirm that the backup/restore option is now available in the WebUI and working!

closed 2023-11-05 by jekader