pixelit-project/PixelIt #322: MQTT message is to long

sp7dpt, 2023-10-29:
I upgrade firmware 2.3.2 to 2.4.3 and mqtt not work. I have error  in log MQTT message is to long. How to fix the problem. In version 2.3.2 everything works good.

foorschtbar, 2023-10-30:
Did you only notice the error message or is there also a problem?

sp7dpt, 2023-10-30:
> Did you only notice the error message or is there also a problem?

There is a message in the LOG and PixelIT does not execute the command.

sp7dpt, 2023-10-30:
In firmware 2.3.2 in LOG i have MQTT_callback: Incomming JSON (Length: 1405), but in firmware 2.4.3 i have 
MQTT_callback: JSON length mismatch! JSON Message to long 
MQTT_callback: Incoming JSON (Topic: Dom/PixelIt/setScreen, Length: 1641/1405)

foorschtbar, 2023-10-30:
How do you send the message? 
Could you share here the message you send? 

Background: In 2.4x, when the input payload and parsed json lenght mismatch, normally the json was to big or is malformed.

sp7dpt, 2023-10-30:
I send message from Home Assistant via MQTT.
For example I tested this message:

{
    "text": {
        "textString": "Pixel it 😀",
        "scrollText": "auto",
        "bigFont": false,
        "centerText": false,
        "scrollTextDelay": 40,
        "hexColor": "#FFFFFF",
        "position": {
            "x": 0,
            "y": 1
        }
    }
}

[image]

foorschtbar, 2023-10-30:
I need the raw mqtt message. Maybe you could try MQTT.fx or similar, connect to your broker and subscribe to `Dom/PixelIt/setScreen` and capture the message.

sp7dpt, 2023-10-30:
I only have MQTT explorer

[image]

sp7dpt, 2023-10-30:
In MQTT explorer I also put string and not working.

[image]

foorschtbar, 2023-10-30:
Thank u. I try to reproduce it in the next days.

smerschjohann, 2023-11-05:
I get the same error. The error happens because of the check on the json length (`length != json.measureLength()`), which I don't really understand. Why is this even checked?

This means that every mqtt json must be minified before sending, otherwise this will never match.

After parsing the JSON file, it does not have to be the same length as the received json. The parser will happily ignore spaces etc.

You can verify this by e.g.

```json
{
  "text": {
    "textString": "Test It :D",
    "bigFont": true,
    "scrollText": "auto",
    "scrollTextDelay": 20,
    "centerText": false,
    "position": {
      "x": 8,
      "y": 1
    },
    "color": {
      "r": 255,
      "g": 255,
      "b": 255
    },
    "hexColor": "#FFFFFF"
  }
}
```

vs.

```json
{"text":{"textString":"Test It :D","bigFont":true,"scrollText":"auto","scrollTextDelay":20,"centerText":false,"position":{"x":8,"y":1},"color":{"r":255,"g":255,"b":255},"hexColor":"#FFFFFF"}}
```

sp7dpt, 2023-11-06:
@smerschjohann I checked, it actually works after proper formatting of the payload, but it is very awkward to write.

@foorschtbar Will this be fixed in the firmware or should I correct formating of all automations?

foorschtbar, 2023-11-06:
Please test the [2.4.6-beta](https://github.com/pixelit-project/PixelIt/releases/tag/2.4.6-beta) release and give us feedback

sp7dpt, 2023-11-06:
> Please test the [2.4.6-beta](https://github.com/pixelit-project/PixelIt/releases/tag/2.4.6-beta) release and give us feedback

I tested the firmware 2.4.6-beta and everything works fine

foorschtbar, 2023-11-06:
thank u. I close this issue and hope #323 and #218 will also be fixed with this.

@smerschjohann fell free to reopen you not working for you

closed 2023-11-06 by foorschtbar

smerschjohann, 2023-11-08:
@foorschtbar thanks a lot, it works for me as well :)