pixelit-project/PixelIt #323: Unable to turn off LEDs

jimmyeao, 2023-10-29:
After the latest update, my Homeassistant service to turn off the display no longer works? (Integrated via MQTT)

jimmyeao, 2023-10-29:
to add, using HA to manually set the state, I can turn it off:
[image]

I dont have this issue with any other MQTT devices, they turn on and off as expected

jimmyeao, 2023-10-29:
rolled back to 2.3.2, fixes it for now

gaetanars, 2023-10-30:
Same issue here on D1 Mini

foorschtbar, 2023-10-30:
I dont use HA. Are you firm with mqtt? I need the mqtt payload you send to turn on/off the leds. With node red the sleep mode works without problems..

gaetanars, 2023-10-30:
The MQTT Payload is available throught homeassistant discovery that is define here : https://github.com/pixelit-project/PixelIt/blob/e6865d9c2310d2aac153e26a082d9608bc8b9071/src/PixelIt.ino#L2943-L2984

foorschtbar, 2023-11-04:
Hey guys, you all have Hassio running?

jimmyeao, 2023-11-05:
I do...

> Hey guys, you all have Hassio running?

jimmyeao, 2023-11-05:
When i try and turn off from Hassio, it sends 
setScreen = {'sleepMode': true}
[image]

But doesnt turn off the screen

gaetanars, 2023-11-05:
The error in the pixels is : 
```
[2023-11-05T18:52:32] MQTT_callback: JSON length mismatch! JSON Message to long :(
[2023-11-05T18:52:32] MQTT_callback: Incoming JSON (Topic: Haus/PixelIt/setScreen, Length: 19/18) 
```

foorschtbar, 2023-11-06:
Please test the [2.4.6-beta](https://github.com/pixelit-project/PixelIt/releases/tag/2.4.6-beta) release and give us feedback

jimmyeao, 2023-11-06:
The Beta fixes it for me :)

> Please test the [2.4.6-beta](https://github.com/pixelit-project/PixelIt/releases/tag/2.4.6-beta) release and give us feedback

jimmyeao, 2023-11-06:
Will there be a further fix for brightness?

foorschtbar, 2023-11-06:
Ups, i closed the wrong issue. I will check the brightness issue soon.

foorschtbar, 2023-11-06:
i see the problem. i optimized the code and sleep mode on/off cant combined with other options now. The brightness will be restored when sleep mode ended. To set a new brightness, you could send them alone. Do you need the option to send end sleepmode and set a new brightness?

jimmyeao, 2023-11-06:
> i see the problem. i optimized the code and sleep mode on/off cant combined with other options now. The brightness will be restored when sleep mode ended. To set a new brightness, you could send them alone. Do you need the option to send end sleepmode and set a new brightness?

Yes, the way Hassio send the command (for some reason) is
`setScreen = {'sleepMode': false, 'brightness': 201} `regardless of the sleepmode, its sending them together

foorschtbar, 2023-11-18:
Please test the [2.4.8-beta](https://github.com/pixelit-project/PixelIt/releases/tag/2.4.8-beta) release and give us feedback.

jimmyeao, 2023-11-20:
> Please test the [2.4.8-beta](https://github.com/pixelit-project/PixelIt/releases/tag/2.4.8-beta) release and give us feedback.

Will test this afternoon, thank you :)

foorschtbar, 2023-11-21:
I close this topic now. Feel free to reopen if the problem with the latest 2.5.0 release still appears.

closed 2023-11-21 by foorschtbar