pixelit-project/PixelIt #337: Ulanzi TC001: Add option to disable battery screen on boot

smart7324, 2024-01-01:
Currently (2.5.0) there is no option to disable the battery status of Ulanzi TC001 on boot. On every boot it shows the battery percentage like this:

[image]

I would like to disable that and show nothing, only the clock. I found an option `bootBatteryScreen` in config.json (this option is not in the UI). If I change it to `false`, the clock doesn't show the battery percentage anymore, but still shows some green and blue pixels on boot.

I'm happy to help, if you need some more information or tests :)

smart7324, 2024-02-01:
The issue is still there.

Metaln00b, 2024-02-01:
The ```bool bootBatteryScreen = VBAT_PIN > 0 ? true : false;``` is true on boot and after the config false, maybe there is a race condition.

The battery bootscreen is inaccurate anyway :D and unnecessary after a no-battery mod. I may be able to help with this.

noiob, 2024-02-14:
the battery screen is only broken if you have logo the boot screen disabled so I'm assuming some stuff doesn't get initialized properly

smart7324, 2024-03-16:
The issue is still there…

StevenKRT, 2025-03-18:
Reopen to be able to set this in the Options > Defaults (true/false)?

foorschtbar, 2025-03-23:
Try [2.5.4-beta](https://github.com/pixelit-project/PixelIt/releases/tag/2.5.4-beta) and give me feedback

StevenKRT, 2025-03-24:
Thank you the function works in the beta. The setting can now be stored in the options, after accepting the configuration, the battery display can now be switched on/off.

By the way, 2.5.4-beta: The update without data deletion failed, but the reinstallation with deletion works with Flashing (Web).

foorschtbar, 2025-04-24:
Fixed with 2.5.4

closed 2025-04-24 by foorschtbar