pixelit-project/PixelIt #342: [FEATURE REQUEST] Cyrillic font (ready-made font and code)

geo-mesh, 2024-01-05:
At the moment, Pixelit does not support the Cyrillic font. I would like to see the Cyrillic alphabet in future firmware releases, for this I created a "UPPERCASE font" of 5px and a "lowercase font" of 4px. I also wrote the HEX-code for these fonts (only the addresses need to be edited). Please let me know if I can now count on Cyrillic support in future projects?

[image]

` 
 # PIXELIT Cyrillic UPPERCASE font
 
    0xE0, 0xA0, 0xE0, 0xA0, 0xA0,             /*[204]   0x80 А */ 
    0xE0, 0x80, 0xE0, 0xA0, 0xC0,             /*[205]   0x81 Б */ 
    0xC0, 0xA0, 0xE0, 0xA0, 0xC0,             /*[206]   0x82 В */ 
    0xE0, 0x80, 0x80, 0x80, 0x80,             /*[207]   0x83 Г */ 
    0x60, 0x50, 0x50, 0x50, 0xE8,             /*[208]   0x84 Д */
    0xE0, 0x80, 0xC0, 0x80, 0xE0,             /*[209]   0x85 Е */
    0xA0, 0x00, 0xE0, 0x80, 0xC0, 0x80, 0xE0, /*[210]   0xF0 Ё */
    0xA8, 0xA8, 0x70, 0xA8, 0xA8,             /*[211]   0x86 Ж */
    0xC0, 0x20, 0x60, 0x20, 0xE0,             /*[212]   0x87 З */
    0xA0, 0xA0, 0xA0, 0xA0, 0xC0,             /*[213]   0x88 И */
    0x40, 0x00, 0xA0, 0xA0, 0xA0, 0xA0, 0xC0, /*[214]   0x89 Й */
    0xA0, 0xA0, 0xC0, 0xA0, 0xA0,             /*[215]   0x8A К */
    0x60, 0xA0, 0xA0, 0xA0, 0xA0,             /*[216]   0x8B Л */
    0x88, 0xD8, 0xA8, 0x88, 0x88,             /*[217]   0x8C М */
    0xA0, 0xA0, 0xE0, 0xA0, 0xA0,             /*[218]   0x8D Н */
    0xE0, 0xA0, 0xA0, 0xA0, 0xE0,             /*[219]   0x8E О */
    0xE0, 0xA0, 0xA0, 0xA0, 0xA0,             /*[220]   0x8F П */
    0xE0, 0xA0, 0xE0, 0x80, 0x80,             /*[221]   0x90 Р */
    0xE0, 0x80, 0x80, 0x80, 0xE0,             /*[222]   0x91 С */
    0xE0, 0x40, 0x40, 0x40, 0x40,             /*[223]   0x92 Т */
    0xA0, 0xA0, 0xE0, 0x40, 0x40,             /*[224]   0x93 У */
    0xF8, 0xA8, 0xF8, 0x20, 0x20,             /*[225]   0x94 Ф */
    0xA0, 0xA0, 0x40, 0xA0, 0xA0,             /*[226]   0x95 Х */
    0xA0, 0xA0, 0xA0, 0xA0, 0xD0,             /*[227]   0x96 Ц */
    0xA0, 0xA0, 0x60, 0x20, 0x20,             /*[228]   0x97 Ч */
    0xA8, 0xA8, 0xA8, 0xA8, 0xF8,             /*[229]   0x98 Ш */
    0xA8, 0xA8, 0xA8, 0xA8, 0xF4,             /*[230]   0x99 Щ */
    0xC0, 0x40, 0x70, 0x50, 0x70,             /*[231]   0x9A Ъ */
    0x88, 0x88, 0xE8, 0xA8, 0xE8,             /*[232]   0x9B Ы */
    0x80, 0x80, 0xE0, 0xA0, 0xE0,             /*[233]   0x9C Ь */
    0xC0, 0x20, 0x60, 0x20, 0xC0,             /*[234]   0x9D Э */
    0xB8, 0xA8, 0xE8, 0xA8, 0xB8,             /*[235]   0x9E Ю */
    0xE0, 0xA0, 0x60, 0xA0, 0xA0,             /*[236]   0x9F Я */

   
    {135, 8, 5, 4, 0, -5}, /*[204]   0x80 А */
    {140, 8, 5, 4, 0, -5}, /*[205]   0x81 Б */
    {145, 8, 5, 4, 0, -5}, /*[206]   0x82 В */
    {150, 8, 5, 4, 0, -5}, /*[207]   0x83 Г */
    {155, 8, 5, 6, 0, -5}, /*[208]   0x84 Д */
    {160, 8, 5, 4, 0, -5}, /*[209]   0x85 Е */
    {165, 8, 7, 4, 0, -7}, /*[210]   0xF0 Ё */ 
    {170, 8, 5, 6, 0, -5}, /*[211]   0x86 Ж */
    {175, 8, 5, 4, 0, -5}, /*[212]   0x87 З */
    {180, 8, 5, 4, 0, -5}, /*[213]   0x88 И */
    {185, 8, 7, 4, 0, -7}, /*[214]   0x89 Й */ 
    {190, 8, 5, 4, 0, -5}, /*[215]   0x8A К */
    {195, 8, 5, 4, 0, -5}, /*[216]   0x8B Л */
    {200, 8, 5, 6, 0, -5}, /*[217]   0x8C М */
    {205, 8, 5, 4, 0, -5}, /*[218]   0x8D Н */
    {210, 8, 5, 4, 0, -5}, /*[219]   0x8E О */
    {215, 8, 5, 4, 0, -5}, /*[220]   0x8F П */
    {220, 8, 5, 4, 0, -5}, /*[221]   0x90 Р */
    {225, 8, 5, 4, 0, -5}, /*[222]   0x91 С */
    {230, 8, 5, 4, 0, -5}, /*[223]   0x92 Т */
    {235, 8, 5, 4, 0, -5}, /*[224]   0x93 У */
    {240, 8, 5, 6, 0, -5}, /*[225]   0x94 Ф */
    {245, 8, 5, 4, 0, -5}, /*[226]   0x95 Х */
    {250, 8, 5, 5, 0, -5}, /*[227]   0x96 Ц */
    {255, 8, 5, 4, 0, -5}, /*[228]   0x97 Ч */
    {260, 8, 5, 6, 0, -5}, /*[229]   0x98 Ш */
    {225, 8, 5, 7, 0, -5}, /*[230]   0x99 Щ */
    {230, 8, 5, 5, 0, -5}, /*[231]   0x9A Ъ */
    {235, 8, 5, 6, 0, -5}, /*[232]   0x9B Ы */
    {240, 8, 5, 4, 0, -5}, /*[233]   0x9C Ь */
    {245, 8, 5, 4, 0, -5}, /*[234]   0x9D Э */
    {250, 8, 5, 6, 0, -5}, /*[235]   0x9E Ю */
    {255, 8, 5, 4, 0, -5}, /*[236]   0x9F Я */    

[image]

 # PIXELIT Cyrillic lowercase font

    0xC0, 0x60, 0xA0, 0xE0,             /*[237]   0xA0 а */ 
    0x60, 0xC0, 0xA0, 0xE0,             /*[238]   0xA1 б */ 
    0x40, 0xA0, 0xC0, 0xA0, 0xC0,       /*[239]   0xA2 в */ 
    0xE0, 0xA0, 0x80, 0x80,             /*[240]   0xA3 г */ 
    0x60, 0x50, 0x50, 0xE8,             /*[241]   0xA4 д */
    0x60, 0xA0, 0xC0, 0x60,             /*[242]   0xA5 е */
    0xA0, 0x00, 0x60, 0xA0, 0xC0, 0x60, /*[243]   0xF1 ё */
    0xA8, 0x70, 0xA8, 0xA8,             /*[244]   0xA6 ж */
    0x20, 0x40, 0x20, 0xE0,             /*[245]   0xA7 з */
    0xA0, 0xA0, 0xA0, 0xC0,             /*[246]   0xA8 и */
    0x40, 0x00, 0xA0, 0xA0, 0xA0, 0xC0, /*[247]   0xA9 й */
    0xA0, 0xC0, 0xA0, 0xA0,             /*[248]   0xAA к */
    0x60, 0xA0, 0xA0, 0xA0,             /*[249]   0xAB л */
    0x88, 0xD8, 0xA8, 0x88,             /*[250]   0xAC м */
    0xA0, 0xA0, 0xE0, 0xA0,             /*[251]   0xAD н */
    0x40, 0xA0, 0xA0, 0x40,             /*[252]   0xAE о */
    0xE0, 0xA0, 0xA0, 0xA0,             /*[253]   0xAF п */
    0xC0, 0xA0, 0xE0, 0x80,             /*[254]   0xE0 р */
    0x60, 0x80, 0x80, 0x60,             /*[255]   0xE1 с */
    0xE0, 0x40, 0x40, 0x40,             /*[256]   0xE2 т */
    0xA0, 0xE0, 0x20, 0xC0,             /*[257]   0xE3 у */
    0x78, 0xA8, 0xF0, 0x20,             /*[258]   0xE4 ф */
    0xA0, 0x40, 0xA0, 0xA0,             /*[259]   0xE5 х */
    0xA0, 0xA0, 0xA0, 0xD0,             /*[260]   0xE6 ц */
    0xA0, 0xA0, 0x60, 0x20,             /*[261]   0xE7 ч */
    0x88, 0xA8, 0xA8, 0x70,             /*[262]   0xE8 ш */
    0x88, 0xA8, 0xA8, 0x74,             /*[263]   0xE9 щ */
    0xC0, 0x60, 0x50, 0x60,             /*[264]   0xEA ъ */
    0x88, 0xC8, 0xA8, 0xC8,             /*[265]   0xEB ы */
    0x80, 0xC0, 0xA0, 0xC0,             /*[266]   0xEC ь */
    0xC0, 0x20, 0x60, 0x20, 0xC0,       /*[267]   0xED э */
    0x98, 0xA8, 0xE8, 0xB0,             /*[268]   0xEE ю */
    0x60, 0xA0, 0x60, 0xA0,             /*[269]   0xEF я */

    {283, 8, 4, 4, 0, -4}, /*[237]   0xA0 а */
    {287, 8, 4, 4, 0, -4}, /*[238]   0xA1 б */
    {292, 8, 5, 4, 0, -5}, /*[239]   0xA2 в */ 
    {296, 8, 4, 4, 0, -4}, /*[240]   0xA3 г */
    {301, 8, 4, 6, 0, -4}, /*[241]   0xA4 д */
    {305, 8, 4, 4, 0, -4}, /*[242]   0xA5 е */
    {310, 8, 6, 4, 0, -6}, /*[243]   0xF1 ё */
    {315, 8, 4, 6, 0, -4}, /*[244]   0xA6 ж */
    {320, 8, 5, 4, 0, -5}, /*[245]   0xA7 з */
    {325, 8, 4, 4, 0, -4}, /*[246]   0xA8 и */
    {331, 8, 6, 4, 0, -6}, /*[247]   0xA9 й */
    {336, 8, 4, 4, 0, -4}, /*[248]   0xAA к */
    {341, 8, 4, 4, 0, -4}, /*[249]   0xAB л */
    {345, 8, 4, 6, 0, -4}, /*[250]   0xAC м */
    {349, 8, 4, 4, 0, -4}, /*[251]   0xAD н */
    {353, 8, 4, 4, 0, -4}, /*[252]   0xAE о */
    {358, 8, 4, 4, 0, -4}, /*[253]   0xAF п */
    {363, 8, 4, 4, 0, -4}, /*[254]   0xE0 р */
    {367, 8, 4, 4, 0, -4}, /*[255]   0xE1 с */
    {371, 8, 4, 4, 0, -4}, /*[256]   0xE2 т */
    {376, 8, 4, 4, 0, -4}, /*[257]   0xE3 у */
    {380, 8, 4, 6, 0, -4}, /*[258]   0xE4 ф */
    {384, 8, 4, 4, 0, -4}, /*[259]   0xE5 х */
    {388, 8, 4, 5, 0, -4}, /*[260]   0xE6 ц */
    {392, 8, 4, 4, 0, -4}, /*[261]   0xE7 ч */
    {397, 8, 4, 6, 0, -4}, /*[262]   0xE8 ш */
    {371, 8, 4, 7, 0, -4}, /*[263]   0xE9 щ */
    {376, 8, 4, 5, 0, -4}, /*[264]   0xEA ъ */
    {380, 8, 4, 6, 0, -4}, /*[265]   0xEB ы */
    {384, 8, 4, 4, 0, -4}, /*[266]   0xEC ь */
    {388, 8, 5, 4, 0, -5}, /*[267]   0xED э */
    {392, 8, 4, 6, 0, -4}, /*[268]   0xEE ю */
    {397, 8, 4, 4, 0, -4}, /*[269]   0xEF я */`

jekader, 2024-01-07:
Thanks for your contribution! The offsets should continue from the main PixelItGlyphs font and then UTF letter assignments can be made in the Utf8ToAscii helper function

geo-mesh, 2024-01-07:
> Thanks for your contribution! The offsets should continue from the main PixelItGlyphs font and then UTF letter assignments can be made in the Utf8ToAscii helper function. You can search the code for 😀 to find an example of how the mapping is done for this emoji.
> 
> For Russian letters you provided, we'll need to map Unicode symbols `U+0410` .. `U+044F` according to https://en.wikipedia.org/wiki/List_of_Unicode_characters#Cyrillic
> 
> By the way, did you create this font yourself or it is taken from some permissive source?
> 
> I can take a look later and create the mapping (since it's linear it can be done in a single `case` statement similar to Latin1 Supplement)

I created it myself and wrote the HEX code for it. At the moment, I have changed the encoding to UTF-16 and edited the glyphs settings. Does it look right now? Can you check it out? If not, write about it and I will fix it.

geo-mesh, 2024-01-08:
```
const uint8_t PixelItBitmaps[] PROGMEM = {
    0xC0, 0x40, 0x20, 0x40, 0xC0,       // [ 93 | 125 | 0x7D] braceright (0x7D)
    0x60, 0xC0,                         // [ 94 | 126 | 0x7E] asciitilde (0x7E)
                                       // Basic Cyrillic (Decimal 1025-1111)
    0xA0, 0x00, 0xE0, 0x80, 0xC0, 0x80, 0xE0, // [95 | 1025 | 0x401] Ё (0x401)
    0x60, 0x80, 0xC0, 0x80, 0x60,             // [96 | 1028 | 0x404] Є (0x404)
    0x80, 0x80, 0x80, 0x80, 0x80,             // [97 | 1030 | 0x406] І (0x406)
    0xA0, 0x00, 0x40, 0x40, 0x40, 0x40, 0x40, // [98 | 1031 | 0x407] Ї (0x407)
    0xE0, 0xA0, 0xE0, 0xA0, 0xA0,             // [99 | 1040 | 0x410] А (0x410)

const GFXglyph PixelItGlyphs[] PROGMEM = {
    {411, 8, 5, 4, 0, -5}, // [ 93 | 125 | 0x7D] braceright (0x7D)
    {416, 8, 2, 4, 0, -5}, // [ 94 | 126 | 0x7E] asciitilde (0x7E)
                           // Glyphs Basic Cyrillic (Decimal 1025-1111)
    {418, 8, 7, 4, 0, -7}, // [95 | 1025 | 0x401] Ё (0x401)
    {425, 8, 5, 4, 0, -5}, // [96 | 1028 | 0x404] Є (0x404)
    {430, 8, 5, 2, 0, -5}, // [97 | 1030 | 0x406] І (0x406)
    {435, 8, 7, 4, 0, -7}, // [98 | 1031 | 0x407] Ї (0x407)
    {442, 8, 5, 4, 0, -5}, // [99 | 1040 | 0x410] А (0x410)
```
[jekader](https://github.com/jekader)
I tried to test my font and added the first 5 lines of my Cyrillic font code. But my symbols  «Ё, Є, І,  Ї, А» are not displayed, instead I get an empty screen. I also set up the "Glyphs" offset according to the continuation of the line code [94]..
Also in the file " Tools.h " via "Utf8ToAscii" I added the following code:

```
case 1025 ... 1040: // Basic Cyrillic (Decimal 1025-1111)
	result = ascii;
	break;
```
But all to no avail.

Could you tell me what can be done to make it work? I'm not an expert, but I figured out how the pixel font is encoded in "HEX", what Decimal and ASCII are. And I really want my font to finally display on my pixel clock. I also spend a lot of time to redo the code of my Cyrillic font for the Pixelit project.

Thanks for any information.

jekader, 2024-01-09:
Hi, the range for the glyphs in UTF-8 should be D081 (Ё U+0401) to D197 (ї U+0457)

here's a placeholder for the contiguous range А..я which just prints question marks instead of the glyphs:

```
	case 0xD090 ... 0xD0BF: // Russian Cyrillic (Decimal 206-253)
		result = 0x3F;
		break;
	case 0xD180 ... 0xD18F: // Russian Cyrillic (Decimal 254-269)
		result = 0x3F;
		break;
```

What is left is to add the offset (0x90 becomes 0xEE and so on) - I am also working on a patch, will let you know the right math once I have it working. For other non-contiguous symbols we'll have to add them individually or draw the missing ones

jekader, 2024-01-09:
Some more thoughts after playing around:
1) the glyph number needs to be changed on the font object when updating the font to enable new letters 
2) setting the max number higher than 0xFF breaks the screen for me
3) there's already 201 glyphs and the limit is 224 which is just 13 more symbols
4) the pixelit font is 5px tall but some of the new letters are 7 pixels tall
5) newly provided Cyrillic letters have a different drawing style than existing Latin matching ones (ABC for instance) - this might need adjustment once we figure out the rendering issue
6) some offsets are off starting with з when previewing via https://tchapi.github.io/Adafruit-GFX-Font-Customiser/

With this, I was able to show some of the chars successfully:
[image]

So besides the minor bugs in the font itself I thing the issue with glyph limits is the biggest. I don't have much experience with the GFX Library - maybe @foorschtbar can comment on this. The `last` attribute is `uint16_t` according to the [reference](https://adafruit.github.io/Adafruit-GFX-Library/html/struct_g_f_xfont.html) so it should be able to go higher than 256. The font customizer also has no problem displaying all the glyphs.

jekader, 2024-01-12:
UPD: as a temporary hack, since we're stuck with 1-byte encodings, I've hacked up a patch that replaces latin extended glyphs with your font - it can be found in this branch:
https://github.com/jekader/PixelIt/tree/cyrillic-font

This only covers the contiguous Russian alphabet mapping in `Utf8ToAscii` at the moment for simplicity. I had issues with glyphs after lowercase `ж` as it is reported as 4 bytes in size while the image looks larger. Offsets and alignment need fixing. All uppercase glyphs work fine (I've trimmed down everything to max 6 pixels high).

Feel free to build this branch and improve on it, should give you the result you need while we figure out a proper solution if one is even possible for embedded devices.

jekader, 2024-01-13:
I've added extended Cyrillic glyphs to my branch. Looks like Ґ and ґ were missing from your font so I've created those and added them as well. Now all East-Slavic Cyrillic languages should fully work.

Looking at the patch, seems like having a separate repo which would rebase with each PixelIt release would be a reasonable solution: whoever needs a PixelIt with Cyrillic support can just install the relevant binary. WDYT?

geo-mesh, 2024-01-15:
[jekader](https://github.com/jekader) 
Thank you for your feedback and your work! Sorry for not responding earlier. 
At some point, I realized that it was not enough just to add a HEX code for Cyrillic to make it work and my hands dropped. 
As I understand it, you created a fork of the project and instead of extended Latin characters, you added Cyrillic for the experiment?
I still do not understand how the function works when entering a character from the keyboard for subsequent transmission to Utf8ToAscii and 1-byte encodings. It is configured in the src/Tools.h file ?

jekader, 2024-01-16:
Yes, correct. `Tools.h` has a `switch` statement which loops through user input chars in UTF-8 format and converts them to a symbol position in the font which is 0..255

closed 2024-02-24 by github-actions[bot]