pixelit-project/PixelIt #348: Node-red Warnung

totocotonio, 2024-04-01:
Hi,
seit dem neusten update nod-red habe ich folgende Warnung.

Node-RED:  1 Apr 15:02:38 - [warn] Context f1ca915489992660:3ae677fd.42e558  contains a circular reference that cannot be persisted
--

node-red.1 | 2024-04-01 15:02:38.750 | warn | Node-RED: 1 Apr 15:02:38 - [warn] Context b22d9ab5.1a84c8:3ae677fd.42e558 contains a circular reference that cannot be persisted

Diese Meldung kommt aus dem Flow der Matrix.

So wie ich das Interpretiere, ist das eine function,  die die Temperatur weiter gibt, denn wenn ich die funktion deaktiviere ist die Fehlermeldung weg.

Das hier steht in der function.

const data=Number(msg.payload).toFixed();
msg.payload=data+' °C';
return msg;

Das hat immer funktioniert.

Kann mir da einer helfen?

foorschtbar, 2024-04-01:
Mach mal aus dem 'const' ein 'var'

totocotonio, 2024-04-01:
Hi,
brachte keine Abhilfe!

foorschtbar, 2024-04-02:
Dann kommt das aus von einer anderen stelle. Wenn du die Funktion deaktivierst, gibt sie auch an die nachfolgenden nodes keine daten mehr. Dort könnte auch der Fehler liegen.