pixelit-project/PixelIt #354: Display switches automatically on

hermannw, 2024-06-24:
Hi guys,
I am using the Node-Red example and I added the following:
[image]
The code in the function nodes:
Display:
```
var night = msg.payload;

if (night == "on")
{
    msg.sleepMode = true;
}
else
{
    msg.sleepMode = false;
}
node.status({ fill: "yellow", shape: "ring", text: "SleepMode is " + msg.sleepMode });

return msg;
```
Display ON:
```
msg.sleepMode = false;
node.status({ fill: "yellow", shape: "ring", text: "SleepMode is " + msg.sleepMode});
return msg;
```
Display OFF
```
msg.sleepMode = true;
node.status({ fill: "yellow", shape: "ring", text: "SleepMode is " + msg.sleepMode});
return msg;
```

When night mode is activated automatically or manually and either the clock or the date is shown at that moment, Pixelit goes to sleep mode (zzz) but a few seconds later it comes back on and keeps showing only the clock and date. 
I am using big fonts. When the clock comes back up, it shows small and bold font for a few seconds and the it switches back to big font.
PixeIt Version 2.5.0

Any ideas why this happens?

Hermann

closed 2024-08-01 by github-actions[bot]

Topper76000, 2024-11-04:
I have the same problem on 2.5.3 and wemos D1 Mini.

foorschtbar, 2024-11-05:
Please export your PixelIt configuration file (Backup & Restore), Remove passwords and share it here

Topper76000, 2024-11-05:
This is my PixelIt config:

{"version":"2.5.3",
"isESP8266":true,
"temperatureUnit":0,
"matrixBrightnessAutomatic":true,
"mbaDimMin":5,
"mbaDimMax":255,
"mbaLuxMin":0,
"mbaLuxMax":10000,
"matrixBrightness":0,
"matrixType":1,
"note":"",
"hostname":"PixelIt-1755203",
"matrixTempCorrection":"default",
"ntpServer":"de.pool.ntp.org",
"clockTimeZone":1,
"clockColor":"#2af10f",
"clockSwitchAktiv":true,
"clockSwitchSec":175,
"clock24Hours":true,
"clockDayLightSaving":true,
"clockWithSeconds":false,
"clockAutoFallbackActive":true,
"clockAutoFallbackTime":15,
"clockAutoFallbackAnimation":4,
"clockDateDayMonth":true,
"clockDayOfWeekFirstMonday":true,
"clockBlinkAnimated":true,
"clockLargeFont":false,
"clockFatFont":true,
"clockDrawWeekDays":false,
"scrollTextDefaultDelay":100,
"bootScreenAktiv":true,
"bootBatteryScreen":false,
"bootSound":false,
"mqttAktiv":true,
"mqttUser":"XXXXXXX",
"mqttPassword":"XXXXXXX",
"mqttServer":"192.168.1.30",
"mqttMasterTopic":"pixelit/",
"mqttPort":1883,
"mqttUseDeviceTopic":true,
"mqttHAdiscoverable":true,
"luxOffset":0,
"temperatureOffset":0,
"humidityOffset":0,
"pressureOffset":0,
"gasOffset":0,
"dfpRXpin":"Pin_D7",
"dfpTXpin":"Pin_D8",
"onewirePin":"Pin_D1",
"SCLPin":"Pin_D1",
"SDAPin":"Pin_D3",
"btn0Pin":"Pin_D0",
"btn0PressedLevel":0,
"btn0Enabled":false,
"btn0Action":2,
"btn1Pin":"Pin_D4",
"btn1PressedLevel":0,
"btn1Enabled":false,
"btn1Action":1,
"btn2Pin":"Pin_D5",
"btn2PressedLevel":0,
"btn2Enabled":false,
"btn2Action":0,
"ldrDevice":"GL5516",
"ldrPulldown":10000,
"ldrSmoothing":0,
"initialVolume":10,
"sendTelemetry":true,
"checkUpdateScreen":true}

Thanks

hermannw, 2024-11-07:
And here is mine. In the mean time I updated to 2.5.3 but the problem is still the same.
```
{
  "version": "2.5.0",
  "isESP8266": true,
  "temperatureUnit": 0,
  "matrixBrightnessAutomatic": false,
  "mbaDimMin": 10,
  "mbaDimMax": 80,
  "mbaLuxMin": 0,
  "mbaLuxMax": 2,
  "matrixBrightness": 60,
  "matrixType": 1,
  "note": "",
  "hostname": "",
  "matrixTempCorrection": "default",
  "ntpServer": "de.pool.ntp.org",
  "clockTimeZone": -6,
  "clockColor": "#3E5AC9",
  "clockSwitchAktiv": true,
  "clockSwitchSec": 7,
  "clock24Hours": true,
  "clockDayLightSaving": true,
  "clockWithSeconds": false,
  "clockAutoFallbackActive": false,
  "clockAutoFallbackTime": 30,
  "clockAutoFallbackAnimation": 1,
  "clockDateDayMonth": true,
  "clockDayOfWeekFirstMonday": false,
  "clockBlinkAnimated": true,
  "clockLargeFont": true,
  "clockFatFont": false,
  "clockDrawWeekDays": true,
  "scrollTextDefaultDelay": 200,
  "bootScreenAktiv": true,
  "bootBatteryScreen": false,
  "bootSound": false,
  "mqttAktiv": true,
  "mqttUser": "",
  "mqttPassword": "",
  "mqttServer": "192.168.1.120",
  "mqttMasterTopic": "pixelit/",
  "mqttPort": 1883,
  "mqttUseDeviceTopic": true,
  "mqttHAdiscoverable": true,
  "luxOffset": 0,
  "temperatureOffset": 0,
  "humidityOffset": 0,
  "pressureOffset": 0,
  "gasOffset": 0,
  "dfpRXpin": "Pin_D7",
  "dfpTXpin": "Pin_D8",
  "onewirePin": "Pin_D1",
  "SCLPin": "Pin_D1",
  "SDAPin": "Pin_D3",
  "btn0Pin": "Pin_D0",
  "btn0PressedLevel": 0,
  "btn0Enabled": false,
  "btn0Action": 2,
  "btn1Pin": "Pin_D4",
  "btn1PressedLevel": 0,
  "btn1Enabled": false,
  "btn1Action": 1,
  "btn2Pin": "Pin_D5",
  "btn2PressedLevel": 0,
  "btn2Enabled": false,
  "btn2Action": 0,
  "ldrDevice": "GL5549",
  "ldrPulldown": 10000,
  "ldrSmoothing": 0,
  "initialVolume": 10,
  "sendTelemetry": false,
  "checkUpdateScreen": true
}```

foorschtbar, 2024-11-07:
> This is my PixelIt config:
> 
> ```json
> {"version":"2.5.3",
> "isESP8266":true,
> "temperatureUnit":0,
> "matrixBrightnessAutomatic":true,
> "mbaDimMin":5,
> "mbaDimMax":255,
> "mbaLuxMin":0,
> "mbaLuxMax":10000,
> "matrixBrightness":0,
> "matrixType":1,
> "note":"",
> "hostname":"PixelIt-1755203",
> "matrixTempCorrection":"default",
> "ntpServer":"de.pool.ntp.org",
> "clockTimeZone":1,
> "clockColor":"#2af10f",
> "clockSwitchAktiv":true,
> "clockSwitchSec":175,
> "clock24Hours":true,
> "clockDayLightSaving":true,
> "clockWithSeconds":false,
> "clockAutoFallbackActive":true,
> "clockAutoFallbackTime":15,
> "clockAutoFallbackAnimation":4,
> "clockDateDayMonth":true,
> "clockDayOfWeekFirstMonday":true,
> "clockBlinkAnimated":true,
> "clockLargeFont":false,
> "clockFatFont":true,
> "clockDrawWeekDays":false,
> "scrollTextDefaultDelay":100,
> "bootScreenAktiv":true,
> "bootBatteryScreen":false,
> "bootSound":false,
> "mqttAktiv":true,
> "mqttUser":"XXXXXXX",
> "mqttPassword":"XXXXXXX",
> "mqttServer":"192.168.1.30",
> "mqttMasterTopic":"pixelit/",
> "mqttPort":1883,
> "mqttUseDeviceTopic":true,
> "mqttHAdiscoverable":true,
> "luxOffset":0,
> "temperatureOffset":0,
> "humidityOffset":0,
> "pressureOffset":0,
> "gasOffset":0,
> "dfpRXpin":"Pin_D7",
> "dfpTXpin":"Pin_D8",
> "onewirePin":"Pin_D1",
> "SCLPin":"Pin_D1",
> "SDAPin":"Pin_D3",
> "btn0Pin":"Pin_D0",
> "btn0PressedLevel":0,
> "btn0Enabled":false,
> "btn0Action":2,
> "btn1Pin":"Pin_D4",
> "btn1PressedLevel":0,
> "btn1Enabled":false,
> "btn1Action":1,
> "btn2Pin":"Pin_D5",
> "btn2PressedLevel":0,
> "btn2Enabled":false,
> "btn2Action":0,
> "ldrDevice":"GL5516",
> "ldrPulldown":10000,
> "ldrSmoothing":0,
> "initialVolume":10,
> "sendTelemetry":true,
> "checkUpdateScreen":true}
> ```
> 
> Thanks

I used your configuration and the sleepmode works without any problems - now over 15min without interruption

- Please check your PixelIt Log
- Please check your flow