pixelit-project/PixelIt #359: Request: use WiFiManager v2.0.17

jekader, 2024-08-12:
Currently, the esp32 target is using WiFiManager v2.0.15-rc.1 while esp8266 is stuck with 0.16.0 for some reason.
The library supports both platforms officially and there's been a new release not that long ago:
https://github.com/tzapu/WiFiManager/releases/tag/v2.0.17

The proposal is to use the latest version across all targets.

I don't have all board types on my hands to test the changes so want to discuss it here first. Is there a reason for an ancient version of WiFiManager to be used for esp8266?

foorschtbar, 2024-08-12:
No special reason, the "old" just works. If someone tests the new lib, we could upgrade.

o0shojo0o, 2024-08-14:
I don't have all the targets available and therefore can't test it.

jekader, 2024-08-14:
It would be enough to test this on a bare esp8266 without even setting up the clock part: as long as it's able to set up the initial connection and raise wifi on boot we are good.

Will look through my drawers to see if I can find a devboard, otherwise they're dirt cheap to order.

foorschtbar, 2024-08-29:
Thanks for the tests so far. The only thing I could do (I lack the time), I can open you a develop-3.0 branch and test the builds on my ESP8266 PixelIts. 

The warning makes sense as sprintf return a null terminated string (the 7th byte) and `hexArray` appears to be only 6 bytes long. If `hexArray` is used as a string in an output, it should be one byte longer to account for this

jekader, 2024-08-29:
Issues sorted out, PR #366 out for review and tested on several of my devices.

closed 2024-08-29 by jekader

foorschtbar, 2024-08-29:
Thank u. I build and released it as beta release 3.0.0 and will test it also the next days :)

PR #366 (2024-08-29): Upgrade SDK for esp8266, use common library versions for all targets
This PR fixes #359 which required also bumping the SDK version for esp8266 to the latest version supported by PlatformIO and adjusting the code base accordingly.

Notable changes:
* `espressif8266 4.2.1` core used
* `EspSoftwareSerial 8.2.0` used for all targets
* `WiFiManager v2.0.17` used for all targets
* `Hash_tng` used for all targets since the code is mostly identical to `mr-glt/SHA-1`
* `ColorConverter` removed due to bugs, `RgbToHex` and `HexToRgb` functions re-created locally
* `webSocket.sendTXT` invocations fixed d [... cut]