pixelit-project/PixelIt #371: Black color wtch digit on 2.5.2 Ulanzi

JediMax27, 2024-10-05:
Hi. Now upgrade Ulanzi 001 watch to 2.5.5. version. But I can't sen any color fot wach digit. When I change it to any color, it steel black after reboot.

Webranger2, 2024-10-05:
I have the same problem

oxivanisher, 2024-10-05:
I have the same issue on my D1 mini version. It is always defaulting to color black and stays black. During boot I can see the Pixel IT logo and shortly the days "lines" at the bottom, so the LEDs are generally working.

rliegmann, 2024-10-05:
Hello everyone, 
i was able to recreate the error.
Can you please take a look at the clock color settings? Is the color code #000000 stored here?
If so, can you change this and see if the time is then given?

Greetings rliegmann

oxivanisher, 2024-10-05:
I tried this, but after the reboot it does on applying the settings, it will go back to black.

foorschtbar, 2024-10-06:
Please go back to 2.5.1. We check the reason at the moment.

foorschtbar, 2024-10-09:
Fixed with https://github.com/pixelit-project/PixelIt/releases/tag/2.5.2 (b96b63a46c339c9a9bfa2b0d64a3a4e9f74936ed)

closed 2024-10-09 by foorschtbar

oxivanisher, 2024-10-09:
> Fixed with https://github.com/pixelit-project/PixelIt/releases/tag/2.5.2 ([b96b63a](https://github.com/pixelit-project/PixelIt/commit/b96b63a46c339c9a9bfa2b0d64a3a4e9f74936ed))

Can confirm on my D1 mini 👍 thanks!