pixelit-project/PixelIt #376: PixelIt 2.5.3 with Ulanzi

JediMax27, 2024-10-13:
Today I install PixelIt 2.5.3 on Ulanzi clock. I have 2 problem with it
1. I try to upgrade firmware from 2.5.1 (use https://pixelit-project.github.io/firmware.html) with no success. It working only when I flash it with "first install" option.
2. when Ulanzi connect to power from usb, on web interface I see too big Luminance (like 623586 lux.) But if I deconnect Usb, luminance is somthing like 1114 lux

foorschtbar, 2024-10-13:
1. The web updater is only for "fresh" installations. Normal updates could done on the WebUI of the PixelIT under `Update & Backup` on the left side.
2. Do you had the same behavior with 2.5.1?

JediMax27, 2024-10-13:
1. Ok, thank you
2. As I remember, no

foorschtbar, 2024-10-13:
Please downgrade to 2.5.1 check this for us :)

Jurich2004, 2024-10-21:
I have PixelIT 2.5.1 on ESP8266_d1_mini and I try to update via WebUI of the PixelIT under Update & Backup to version 2.5.3. After pressing the update button the firmware device freezes and reboot helps. Update does not happen, still version 2.5.1.
Similarly, I updated PixelIT 2.5.1 on ESP32_d1_mini32 and everything was fine there. The update happened.

foorschtbar, 2024-10-26:
We had over 250 updates without issues, please try it via cable (web flasher). it might be a problem with your device or a corrupt file.

closed 2024-10-26 by foorschtbar