pixelit-project/PixelIt #382: DST 2 days early

sjthespian, 2024-11-02:
My pixel seems to have decided that DST came early this year. Friday morning at 2am (my spouse was up) the clock went back an hour. I'm not sure where it gets its DST data for Northern California in the US, but it looks like it is a couple of days off. It should roll the clocks back tomorrow morning, 2024-Nov-3, not 2024-Nov-1.

I've tried toggling DST on and off again in the settings which doesn't change the time. The only workaround is to change the timezone offset. 

| ---- | ---- |
| PixelIt Version | 2.5.3 |
| Hostname | pixelit01 |
| Device-ID | PixelIt-11560495 |
| Build Section | ESP8266_d1_mini |
| Free Sketch Space | 2.6 MB |
| Wifi RSSI | -58 dBm |
| Wifi quality | 84 % |
| Free heap | 16.1 kB |
| Current Brightness | 5 |
| Sketch size | 498.0 kB |
| ChipID | 11560495 |
| CPU freq. | 80 MHz |
| Sleep mode | Off |
| Uptime | 1d 1h 7m 54s |
| Reset reason | Software/System restart |
| Matrix size | 32 x 8 (cols x rows) |

sjthespian, 2025-03-11:
I know this issue auto-closed, but i’m seeing the same issue this year. DST started yesterday and the pixelit is an hoir off.

foorschtbar, 2025-03-11:
Looks like a problem with the used TimeLib. There a some other issues in the Repo. We need to switch it. I reopen this. Thank u.

closed 2025-06-01 by github-actions[bot]